ArduPilot/apm_planner #917: Radio config: does not exit command mode.

AndKe, 2016-03-15:
To reproduce: 
connect a telemetry radio, and read settings more than once, you will see that it fail every-second time.
it's also prossible to leave the radio in command mode, so telemetry link won't be successful.

The radio is correctly switched into command mode using "+++"  - but then the same "+++" appears to be tried as exit.
Correct command mode exit is "ATO"

Arne-W, 2017-01-30:
The problem (one of the problems) is the the retry counter - I will fix it in the next days.
Stay tuned...

Arne-W, 2017-02-02:
@AndKe It would be great if you could test my fixes with you radio. For mine it works perfectly now.

AndKe, 2017-02-02:
@Arne-W  It works great. tested with local & remote radio's data.
good job. 

one thing remain...:
(You need only to connect one radio to reproduce)
After radio settings are read, if you unplug it, segfault follows:

```
[20170202 23:05:13.362 INFO ] - local Radio Read Complete success: true
[20170202 23:05:13.369 DEBUG] - readLocalRssi
[20170202 23:05:13.378 DEBUG] - Radio readData currentline: "ATI7\r\n"
[20170202 23:05:13.393 DEBUG] - Radio readData currentline: "L/R RSSI: 55/0  L/R noise: 77/0 pkts: 0  txe=0 rxe=0 stx=0 srx=0 ecc=0/0 temp=-276 dco=0\r\n"
[20170202 23:05:13.394 DEBUG] - Radio3DRSettings::readRemoteVersionString()
[20170202 23:05:13.394 DEBUG] - remote status: "Reading remote version"
[20170202 23:05:13.397 DEBUG] - Radio readData currentline: "RTI0\r\n"
[20170202 23:05:13.875 DEBUG] -  Failed to Read Remote Version
[20170202 23:05:13.876 INFO ] - remote Radio Read Complete success: false
[20170202 23:05:13.876 DEBUG] - remote status: "FAILED"
[20170202 23:05:19.526 ERROR] - Crtical Error! "Resource temporarily unavailable"
Segmentation fault (core dumped)

 ```

Arne-W, 2017-02-02:
Cool - Thank you for the fast testing :+1: 
I will check that - seems to be simple to reproduce.

AndKe, 2017-02-02:
a little more data: 

```
[20170202 23:16:38.974 ERROR] - Crtical Error! "Resource temporarily unavailable"

Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x00007ffff3649850 in QIODevice::errorString() const () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
(gdb) backtrace
#0  0x00007ffff3649850 in QIODevice::errorString() const () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#1  0x0000555555a374ca in Radio3DRSettings::handleError(QSerialPort::SerialPortError) ()
#2  0x0000555555b6d47b in Radio3DRSettings::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#3  0x00007ffff374b35e in QMetaObject::activate(QObject*, int, int, void**) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#4  0x00007ffff3e0ac21 in QSerialPort::error(QSerialPort::SerialPortError) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#5  0x00007ffff3e0ec15 in ?? () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#6  0x00007ffff3e0cf40 in QSerialPort::close() () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#7  0x00007ffff3e0cfc3 in QSerialPort::~QSerialPort() () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#8  0x00007ffff3e0cfd9 in QSerialPort::~QSerialPort() () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#9  0x0000555555a375bb in Radio3DRSettings::handleError(QSerialPort::SerialPortError) ()
#10 0x0000555555b6d47b in Radio3DRSettings::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#11 0x00007ffff374b35e in QMetaObject::activate(QObject*, int, int, void**) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#12 0x00007ffff3e0ac21 in QSerialPort::error(QSerialPort::SerialPortError) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#13 0x00007ffff3e101b3 in ?? () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#14 0x00007ffff3e138bd in ?? () from /home/andre/Qt/5.5/gcc_64/lib/libQt5SerialPort.so.5
#15 0x00007ffff5b1c4cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Widgets.so.5
#16 0x00007ffff5b21630 in QApplication::notify(QObject*, QEvent*) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Widgets.so.5
#17 0x00007ffff371d663 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#18 0x00007ffff3772fe5 in ?? () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#19 0x00007fffef0347d7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007fffef034a40 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007fffef034aec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff3772a87 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#23 0x00007ffff371b212 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#24 0x00007ffff3722d3d in QCoreApplication::exec() () from /home/andre/Qt/5.5/gcc_64/lib/libQt5Core.so.5
#25 0x00005555555fb40c in main ()
(gdb) 

 ```

Arne-W, 2017-02-04:
Made a fix...

AndKe, 2017-02-05:
Works great - although I think a focus-stealing popup dialog belongs in the 90's
Maybe: "Link settings"  (com port, and baud rate)  should be changed to red (and be green when ok.)  - and load/write/restore/flash buttons should be grayed out when connection is not made, or failed. 

Anyway: good job on preventing crash. :)

Arne-W, 2017-02-07:
I used a popup cause the "load settings" button also opens a popup if no interface is connected.
I am not sure if its a good idea just to disable the buttons if there is no interface - perhaps it is a little bit too subtle?!
I will try to change the behaviour like you described above - than we can decide...

Arne-W, 2017-02-14:
If you have some time to waste :grinning: you could test [this branch](https://github.com/Arne-W/apm_planner/tree/3dr-Config-changes). The buttons are disabled as long as there is no interface and the link settings will show "None". The error popup windows are removed and the Status line will show the state and the errors. Errors should be red and successful operations shall be green.

AndKe, 2017-02-15:
@Arne-W  works great , very intuitive/foolproof,  should be pulled asap.  :) 
 Good work, thank you.

Arne-W, 2017-02-15:
Nice to hear that you like it - and thanks again for that incredible fast testing :+1: 
There is still some work - clean up, reorganize commits... but this should be done until weekend.
@billbonney I will add this change to #1004 after that you could review and pull it.

billbonney, 2017-02-16:
Ok, i'll get to merging these changes in then

commit d32aa406 (2017-02-17): 3DR Radio Config: Several Bugfixes Fixes #917

link will be only set if changed; added sleep between the remote and local reset (otherwise the remote reset won't be transmitted); corrected setup of remote frequency boxes; filter unused ascii data; disconnect from signals before reconnect; better detection of reset; correced handling of disconnect (won't crash anymore); leave commandmode on close of connection; reset retrycounter on disconnect.
closed 2017-02-17 by billbonney