ArduPilot/apm_planner #918: Graph: "use time on X-Axis" is not an option with Arduplane logs

AndKe, 2016-03-17:
This is related to https://github.com/diydrones/apm_planner/commit/f9fc68b5a64665bdb561692209122a3782b27c4f

Arne-W, 2016-03-17:
The "use time on X-Axis" feature is only available if every row of the log file has a timestamp. This is only supported by dataflash (.bin) logs. It should not make any difference if its a plane or a copter. 
Just send me a log (upload or mail) and I will check it.

AndKe, 2016-03-17:
here is one plane.BIN http://flyfoto-nord.no/tmp/141.BIN

Arne-W, 2016-03-18:
There are 2 problems with this log. The first one (the smaller one) is that the time colum is called "TimeMS" but the algorithm searches for "TimeUS". The second (and bigger) problem is that not all rows have a timestamp. In the Coper logs there are always timestamps which makes the handling straight forward. Creating several exeptions for with different errorhandling for planes and copters is not a good solution I think.
Perhaps the ArduPlane Guys could introduce a timestamp in every row??

AndKe, 2016-03-18:
Thank for looking into it.
yes, there is no local TimeMS for GPS and CAM messages (kind of logical, as
it has GPS time) , but MSG and PM does not have timestamp.
I asked this question in gitter , I guess that with the frequency of GPS
messages being logged, if local ms were added, that alone would grow logs
even more.

As for solution: could you just ignore these messages for x-axis time, just
take next log item ?

billbonney, 2016-03-18:
Interesting every tlog entry is time stamped on receiving in GCS and saved. 

For non time stamped messages in a bin we could just give then the same time as the last message or add 1 to that time. 

It would be good to find a solution

AndKe, 2016-03-18:
newer Arduplane does add local timestamp to GPS and CAM messages, AFAIK no change for MSG/PM , so handling of missing timestamp would help anyway.

Arne-W, 2016-03-18:
Okay - I will try to find a good solution for this... but it might take a little as the generation of artificial is not the only thing also the format description has to be patched to get a consistent datamodel. Furthermore the copter timestamps are in 'us' while the plane timestamps are in 'ms' which affects the scaling and last but not least all message types ('MODE','ERR'...) have to handle different names for the timestamp column....
I will do my best :-)

Arne-W, 2016-03-23:
Made a solution for this :-)

closed 2016-03-27 by billbonney

AndKe, 2016-03-29:
neat, thank you.

Arne-W, 2016-03-29:
As I said before - you are always welcome and nice to hear that you like it! :-)