ArduPilot/apm_planner #920: Graph: "Show MSG" does not work for .tlog files.

AndKe, 2016-03-22:
[2016-03-22 15-05-18.tlog.zip](https://github.com/diydrones/apm_planner/files/184367/2016-03-22.15-05-18.tlog.zip)

please see attached tlog.
it contains:
```
andre@Loke:~$ mavlogdump.py ~/apmplanner2/tlogs/hexcopter/2016-03-22\ 15-05-18.tlog |grep STATUSTEXT
2016-03-22 15:05:18.66: STATUSTEXT {severity : 1, text : APM:Copter V3.3.2-rc2 (1883b1b8)}
2016-03-22 15:05:18.85: STATUSTEXT {severity : 1, text : PX4: 34e1d543 NuttX: 7c5ef883}
2016-03-22 15:05:18.89: STATUSTEXT {severity : 1, text : Frame: HEXA}
2016-03-22 15:05:19.01: STATUSTEXT {severity : 1, text : PX4v2 00220034 3432470E 31323533}
2016-03-22 15:05:33.93: STATUSTEXT {severity : 3, text : PreArm: Throttle below Failsafe}
2016-03-22 15:06:04.85: STATUSTEXT {severity : 3, text : PreArm: Throttle below Failsafe}
2016-03-22 15:06:28.38: STATUSTEXT {severity : 3, text : Arm: Safety Switch}
2016-03-22 15:07:31.25: STATUSTEXT {severity : 3, text : Arm: Safety Switch}
2016-03-22 15:07:38.67: STATUSTEXT {severity : 3, text : Arm: Safety Switch}
2016-03-22 15:07:43.04: STATUSTEXT {severity : 3, text : Arm: Safety Switch}
2016-03-22 15:08:23.13: STATUSTEXT {severity : 1, text : Initialising APM...}
2016-03-22 15:08:23.25: STATUSTEXT {severity : 1, text : Calibrating barometer}
2016-03-22 15:08:24.86: STATUSTEXT {severity : 1, text : barometer calibration complete}
2016-03-22 15:08:24.88: STATUSTEXT {severity : 1, text : GROUND START}

```

but it's not showing up in graph.

Arne-W, 2016-03-29:
With my new PR #923 this should now work in tlogs. 
By the way - your sample log has some errors - which will be detected in this new PR. The time is not linear - it jumps back at the end of the log - strange!

AndKe, 2016-03-29:
Very strange with the time jumping. It's recorded using AP2, so it intrigued me.  (and I did *not* play with the flux capacitor at the time.)
Did a quick dump of the log in terminal (mavlogdump.py)  , and scrolled up'n'down , but could not spot any time problems. (tell where if you have the time at hand...)
the time jump, is it less few seconds ?  - maybe the computer did a NTP sync ?

Arne-W, 2016-03-30:
As the time stamps in the log-file are read from data and not from the host it must be something that affects time on the flight-controller.
You should be able to visualize it by loading your log and plot raw_imu time_usec in index view. I made some screenshots...

[image]
[image]

As you can see in the end the time jumps back and starts to grow again. Perhaps some sort of unplugging / plugging USB connection?!
However if plotted in "time on x-axis" mode the data would show up corrupted as some data from the end of the log would be inserted at the beginning. So the algorithm replaces such "wrong" timestamps with the last valid one which leads to a cut off at the time warp cause all following data-packets will have the same time.
In index view everything should be normal.

Arne-W, 2016-03-30:
While testing I found that some of my tlogs have the same issue. Looks like a FC reset resets the time too - well no suprise - but AP2 does not start a new log - maybe this is an issue in log generation?!

AndKe, 2016-03-30:
Yes, it would be preferred to write new .tlog record new log on reboots,
even preflight reboot.
Not only would it be cleaner, but also correct in those situations where
GCS is left running, or suspended by OS, between mission parts.

commit a2cb5ca3 (2016-04-06): Graph View: Enable STATUSTEXT tlog message to be handeled as MsgMessage. Fixes #920
closed 2016-04-06 by billbonney