ArduPilot/apm_planner #924: .tlog stops loading

AndKe, 2016-04-08:
please see .tlog posted by some user at http://diydrones.com/forum/topics/hexacopter-crash-of-a-break-prop-in-flight?commentId=705844%3AComment%3A2232054

it stops loading at ~30%

Arne-W, 2016-04-09:
I will have a look...

Arne-W, 2016-04-09:
I have tested this tlog with current master and I can't reproduce this issue. The log loads pretty normal.
`Plot Log loading took 2.631 seconds - 1699659 of 1699659 bytes used `
There are only some issues with the timeline due to some reboots of the FC.

dcarpy, 2016-04-13:
I can't reproduce the issue with either 2.0.18 or master.

AndKe, 2016-04-13:
I am sorry, should have confirmed it right after @Arne-W failed to find a problem.
I do analyze some forum logs to help people, and have a lot of them in "Download" folder, apparently there was another log(or log type) that failed to work, because the one I submitted here works for me too.

closed 2016-04-13 by AndKe