ArduPilot/apm_planner #927: Cannot load firmware for pixhack

martinjuju, 2016-04-12:
Hi I cannot load or update the firmware for a pixhack autopilot.
I got the following message in the console
DEBUG 2016-04-12T15:01:06.469 Requesting firmware: "stable" "" 
ERROR 2016-04-12T15:01:06.470 Unknown autopilot in ApmFirmwareConfig::requestFirmwares() "" 

Everything else is working fine with the pixhack in apmplanner.

If I try using Mission Planner everything works fine and I can load the firmware.

AndKe, 2016-04-13:
This https://github.com/ArduPilot/apm_planner/commit/98c08f19d30dcebc366f91e27130a8e6da4de369  was fixed somewhere during rc4 , and before rc5  - 
Please upgrade to 2.0.20 , current version, and I guess it'll work fine.
close this issue if it does.

martinjuju, 2016-04-13:
Updating fixed it. Issue closed, however, the auto-update was broken in version 2.0.19rc4 and this is why I was not on the latest version.

closed 2016-04-13 by martinjuju