ArduPilot/apm_planner #929: Copter-3.4 PID parameter name changes

rmackay9, 2016-04-14:
We've had to change the names of some PID parameters in Copter-3.4. You don't really need to worry about the scaling changes, that's all handled automatically as part of the upgrade.

Old	                New	                       Scaling change (i.e. multiply old value by below value to arrive at new value)
STB_RLL_P	ATC_ANG_RLL_P	none
STB_PIT_P	ATC_ANG_PIT_P	none
STB_YAW_P	ATC_ANG_YAW_P	none
		
RATE_RLL_P	ATC_RAT_RLL_P	0.9 (for x-frame multicopters) or 1.27
RATE_RLL_I	ATC_RAT_RLL_I	1/4500
RATE_RLL_IMAX	ATC_RAT_RLL_IMAX	1/4500
RATE_RLL_D	ATC_RAT_RLL_D	0.9 (for x-frame multicopters) or 1.27
RATE_RLL_FILT_HZ	ATC_RAT_RLL_FILT	none
		
RATE_PIT_P	ATC_RAT_PIT_P	0.9 (for x-frame multicopters) or 1.27
RATE_PIT_I	ATC_RAT_PIT_I	1/4500
RATE_PIT_IMAX	ATC_RAT_PIT_IMAX	1/4500
RATE_PIT_D	ATC_RAT_PIT_D	0.9 (for x-frame multicopters) or 1.27
RATE_PIT_FILT_HZ	ATC_RAT_PIT_FILT	none
		
RATE_YAW_P	ATC_RAT_YAW_P	none
RATE_YAW_I	ATC_RAT_YAW_I	1/4500
RATE_YAW_IMAX	ATC_RAT_YAW_IMAX	1/4500
RATE_YAW_D	ATC_RAT_YAW_D	none
RATE_YAW_FILT_HZ	ATC_RAT_YAW_FILT	none

For reference, here's the issue for the qGroundControl people. https://github.com/mavlink/qgroundcontrol/issues/3152

billbonney, 2016-04-14:
Display the values and 'sliders' etc for the values is easy enough. The challenge is who the extended tuning screen is impacted (See AC3.3 for reference below)

[image]

billbonney, 2016-09-13:
Completed with https://github.com/ArduPilot/apm_planner/commit/2ba7e87a7dd00df299227c3ad92f8d6e993c4c74
 And some other minor commits

closed 2016-09-13 by billbonney