ArduPilot/apm_planner #932: Widget: Info View (Quick) poor choice of colors.

AndKe, 2016-04-18:
Maybe something else than random would be nice?

[image]

Ultimately, it would be nice to have the ability to set colors, and/or set colors based on value.

dcarpy, 2016-05-19:
@AndKe Have you noticed by chance if the colors have been better since commit 68f292e?  Just checking before I make a similar change in AP2DataPlot2D.cpp...

AndKe, 2016-05-19:
True, it's definitely not making crazy choices, mostly by avoiding extreme light/dark colors, but it has a tendency to end up with very similar colors, not sure how good that is for graphs.

[image]

[image]

dcarpy, 2016-05-20:
@AndKe Agreed.  I'm thinking maybe for both I should use the [Golden Ratio](http://martin.ankerl.com/2009/12/09/how-to-create-random-colors-programmatically/).

AndKe, 2016-05-20:
Yes, Golden Ratio looks good, 
Looking at the examples:
[image]

I see that 8 colors are chosen before the pattern repeats with less different colors.
the best solution, at least for graphs, could be to toggle between the settings every 8 colors:
like on the picture above: 
first 8 colors picked from A-H from the upper example, then next 8 colors from A-H on the lower example, then next 8 colors from upper I-P 8 and so on...

commit 25c9d9e1 (2016-05-29): Graph View: Use HSV Color Model Fixes #932

HSV represents color in a way more suited to the human perception of
color. Still uses random colors, however, should work better on both
light and black background.
closed 2016-05-29 by billbonney