ArduPilot/apm_planner #934: Empty "Standard Params" and  "Advanced Params"

martinjuju, 2016-04-21:
Hi
since I installed the latest version 2.0.20 the "Standard Params" and "Advanced Params" tabs in "Config/Tuning" are empty. I can see everything in "Full Parameters list" and the "Extended tuning" tabs.
Tested on two different PCs with Ubuntu 15.10.

martinjuju, 2016-04-21:
I want to add that this is a packaging problem since it is solved if I build it from source

billbonney, 2016-04-21:
The file with the descriptions is also downloaded. It should not matter if it is in the package

We have had similar reports but I haven't been able to reproduce.

martinjuju, 2016-04-21:
Oddly enough I repeated the problems on 3 different Ubuntu 15.10 system using the .deb package provided.
Building my own .deb (using debuilt) and using the deb file on the 3 systems solved the problem.

dcarpy, 2016-04-22:
Is it the same user you build with, run with and installed with?  Just wondering if it's a permissions problem with the parameters file location.
I'm not able to reproduce either.  Looks good on my box. :-)

salomatu, 2016-05-26:
I think I have something fitting to this description. I found that when the parameters list is downloaded, the file contains the 302 response from the server and apparently whichever component is responsible for the download fails to recurse into the proper address of the file.
I use 2.0.20 and tried to revert to older version, but of course that didn't help since the location of the file has changed.
I circumvent this problem by downloading the file manually with wget each and every time I start the software.
Platform is OSX 10.9.5.

billbonney, 2016-05-26:
it's fixed here 5c81f622e212be05b248e0fba4bb8b45c64b609e

Arne-W, 2017-03-29:
As this is fixed on master - I will close it.

closed 2017-03-29 by Arne-W