ArduPilot/apm_planner #940: segfault on loading .BIN logs (not 100% reproducible, but I'd say 40% crash)

AndKe, 2016-04-28:
```
qml: APMToolBar: GRAPHS SELECTED
qml: APMToolBar: clear selected buttons
qml: APMToolBar: FLIGHT DATA unselected
qml: APMToolBar: GRAPHS unselected
qml: APMToolBar: GRAPHS SELECTED
[New Thread 0x7fff6bfff700 (LWP 21683)]
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
DEBUG 2016-04-28T10:17:59.731 "retrieve versionobject from server: http://firmware.ardupilot.org/Tools/APMPlanner/apm_planner_version.json" 
[New Thread 0x7fff6b7fe700 (LWP 21684)]
DEBUG 2016-04-28T10:18:00.006 Downloading: 2795 / 2795 
DEBUG 2016-04-28T10:18:00.007 Downloading: 2795 / 2795 
DEBUG 2016-04-28T10:18:00.007 AutoUpdateCheck::httpFinished() 
[New Thread 0x7fff6affd700 (LWP 21685)]
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
[Thread 0x7fffcd055700 (LWP 21660) exited]
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QObject::connect: No such signal QTableView::currentCellChanged(int,int,int,int)
QObject::connect:  (sender name:   'tableWidget')
QObject::connect:  (receiver name: 'AP2DataPlot2D')
DEBUG 2016-04-28T10:18:12.847 Created AP2DataPlotThread: AP2DataPlotThread(0x4b34c80) 
[New Thread 0x7fffcd055700 (LWP 21689)]
DEBUG 2016-04-28T10:18:12.848 AP2DataPlotThread::run(): Log loading start - 11354711 bytes 
QObject::connect: Cannot queue arguments of type 'QTextBlock'
(Make sure 'QTextBlock' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
 INFO 2016-04-28T10:18:18.289 Plot Log loading took 5.441 seconds - 11354711 of 11354711 bytes used 
[Thread 0x7fffcd055700 (LWP 21689) exited]

Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x00007fffe9fc3a9c in QFontEngineFT::shouldUseDesignMetrics(QFlags<QFontEngine::ShaperFlag>) const () from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
(gdb) 

```

seems to happen regardless of log file, reproduction procedure:
open ap2 , select graph, load .bin

please note that teh "No space left on device" messages spawn also in successful runs, and there is nothing near "no space left" , in any mounted device/RAM.

Arne-W, 2016-04-28:
Hmm - the log loading itself seems to be OK cause the loading thread terminates normally.
It must be something in the further processing.
When the log loads without a crash are there any errors while loading, or does it load without any issues? 
Could you please upload or mail the log - I will try to reproduce it. 60% percent - should be easy to reproduce.

AndKe, 2016-04-28:
please see https://gist.github.com/AndKe/a5876e79e6dc8023964a9b5fd38e67fd
. deleted .ini,  compiled fresh master..

The log shows consecutive crashes, then a successfull opening of the file.

Arne-W, 2016-04-28:
Ahhh great thanks - it is a Copter V3.3.3 binary log.... are you able to reproduce it with an older bin log? Moreover it would be perfect if could send or upload this particular binary log cause I haven't updated my pixhawk and do not have any log from the V3.3.3.

AndKe, 2016-04-28:
The .BIN , https://drive.google.com/open?id=0B9Rr1JTp37U9SnlKZm9kTUEwTjA
Will try with older logs.

AndKe, 2016-04-28:
extra info, spent 5minutes crashing with a log from AC3.3-rc6 , until it
finally loaded - same issue  - it would be sweet to have an command line
parameter til open logs or drag'n'drop :) (having no idea of how much work
that last thing is.)

Arne-W, 2016-04-28:
OK thanks for your effort - I will have a look this evening

Arne-W, 2016-04-28:
I am really sorry but I can't reproduce this issue on current master.
I loaded your log more than 20 times without any crash or segfault. 
So I took a closer look at your apm-planner log and it looks like that you have build an older apm-planner version! I have just compiled the actual master and the beginning of my logfile looks like this:
 

> INFO 2016-04-28T20:22:14.057 QGCCore::initialize() 
>  INFO 2016-04-28T20:22:14.057 Current Build Info 
>  INFO 2016-04-28T20:22:14.057 Git Hash: b144e8b31bdff81822c46d6a81b1c781a9259889 
>  INFO 2016-04-28T20:22:14.057 Git Commit: 2.0.20-2-gb144e8b-DEV 
>  INFO 2016-04-28T20:22:14.057 APPLICATION_NAME: "APM Planner" 
>  INFO 2016-04-28T20:22:14.057 APPLICATION_VERSION: "v2.0.20" 
>  INFO 2016-04-28T20:22:14.057 APP_PLATFORM: ubuntu64 
>  INFO 2016-04-28T20:22:14.057 APP_TYPE: stable 

If you compare version and Git Hash you will see the mismatch!

AndKe, 2016-04-28:
True, the first gdb error submit was on a build from a week ago or so, 
the gist logs (multiple attempts in a row, then success)  - was with today's , current master

AndKe, 2016-04-28:
please see this: https://gist.github.com/AndKe/c49af4f4c97b37fa138ef5b11507dddb
That's a brand new reproduction on current master (this time hash will match)  ~3 crashes then success on the same log file.

Arne-W, 2016-04-29:
All right - the hash and version mismatch was my fault - so sorry - used a cloned version from my repo.
But now I used the original repo an made a fresh build on another ubuntu (14.04) machine and I still can not reproduce those crashes.
The only thing I can do is to create an special version for you and enrich it with debug messages to get some more details about where the crash happens. I think it happens somewhere in [this method].(https://github.com/ArduPilot/apm_planner/blob/master/src/ui/AP2DataPlot2D.cpp#L1367)
If you are interested I will make a special version this evening and inform you when its done - perhaps together we can find the culprit.

AndKe, 2016-04-29:
Yes please.  I can reproduce it at work as well, on an completely different
computer, different logs (same build)
I start to suspect something related to Ubuntu 16.04 - as that's the only
common thing between those computers.  - but it's kind of strange to think
that. :)

Arne-W, 2016-04-29:
I have added some debug output to the thread done method hopefully it helps to isolate the bug.
You will find those changes [here](https://github.com/Arne-W/apm_planner/tree/Segfault_Fix_wip).

AndKe, 2016-04-29:
Thank you.
compiled, opened one of the logs (it worked perfectly)  , closed the graph window - and opened the same log again, it crashed.
The result is here:
https://gist.github.com/AndKe/df5d64ac905cad0862dd0ba5d8660585

Arne-W, 2016-04-29:
Sorry it took that long - I had to eat something first :grin: 
Yes I have seen that there were no log output :confused: It looks like something with the signals goes wrong. I have simplified the done signal from the parser (no parameters anymore) perhaps it prevents the crash but it is not a solution. We will see... just update and try :+1:

AndKe, 2016-04-29:
Hi, here is the result:

```
[New Thread 0x7fff6b7fe700 (LWP 8413)]
DEBUG 2016-04-29T22:44:51.723 Downloading: 2795 / 2795 
DEBUG 2016-04-29T22:44:51.724 Downloading: 2795 / 2795 
DEBUG 2016-04-29T22:44:51.724 AutoUpdateCheck::httpFinished() 
[New Thread 0x7fff6affd700 (LWP 8414)]
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: No space left on device
QObject::connect: No such signal QTableView::currentCellChanged(int,int,int,int)
QObject::connect:  (sender name:   'tableWidget')
QObject::connect:  (receiver name: 'AP2DataPlot2D')
DEBUG 2016-04-29T22:44:59.694 Created AP2DataPlotThread: AP2DataPlotThread(0x4b2ee50) 
[New Thread 0x7fff6a7fc700 (LWP 8420)]
DEBUG 2016-04-29T22:44:59.695 AP2DataPlotThread::run(): Log loading start - 6171805 bytes 
QObject::connect: Cannot queue arguments of type 'QTextBlock'
(Make sure 'QTextBlock' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
[Thread 0x7fffcd055700 (LWP 8390) exited]
 INFO 2016-04-29T22:45:02.628 Plot Log loading took 2.934 seconds - 6171805 of 6171805 bytes used 
[Thread 0x7fff6a7fc700 (LWP 8420) exited]

Thread 1 "apmplanner2" received signal SIGSEGV, Segmentation fault.
0x00007fffe9fc3a9c in QFontEngineFT::shouldUseDesignMetrics(QFlags<QFontEngine::ShaperFlag>) const () from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
(gdb) 
```

sadly , no change, nor clue about what's going on.

Arne-W, 2016-04-29:
Hmmm - this is my last idea for now. I fixed the warnings about the signals just to be sure...
If this does not help I don't have an Idea what to do next - perhaps asking google.... I think we need a real QT-Guru then...

Arne-W, 2016-04-29:
asking google showed some bugs related to QFontEngineFT with random crashes. Someone wrote about memory which is referenced from QFontEngineFT but freed by a terminating thread. So I added a 10 second sleep at the end of the parsing thread - perhaps it helps but its a kind of poking around in the dark.
So sorry that I can't provide a solution for  you :disappointed:

AndKe, 2016-04-30:
No need to apologize, I am very grateful for your work as-is.
you are also onto something. the current version with sleep - fixes the bug.
there is a log : https://gist.github.com/AndKe/59bdfa7279d01a79b6d9079b52e87d24
in short - graphs open reliably every time, but the app will crash if graph is closed shortly after being open.
An great improvement, which I am sure is trivial to work around(delay close , or reduce delay to a minimum).

Yesterday, before sleep, I found a very similar problem also apparently random, for one another application:
https://bugs.kde.org/show_bug.cgi?id=348988  

It seems like this have something to do with multithreading/timing, and only some applications are affected.

I'll need some family-time today, but will later try to reduce the delay to 0.5sec or so, in hope that fixes it as well - if so, that could be the simplest fix, acceptable for master ?

Arne-W, 2016-04-30:
Yeah :smiley: nice to hear that - so we know what happens - perfect! But I think the sleep is really the last way to solve (workaround) the problem. There will be always situations when the sleep won't work - okay the risk of a crash is reduced but it would be better if the fix would always work.
I will create a new branch with the warning fixes and try to find a better solution than sleep perhaps there is one. I will inform you when I found one or became desperate :grin:

Arne-W, 2016-04-30:
That was easy. I made a fast hack - not tested very good. I replaced the the sleep with a semaphore which is set from the threadDone method. So the loader thread will be blocked until the threadDone method has finished. Hope this fixes the issue too.
Until now I didn't create a new branch so you could test it by pulling the old one. I will create a pull request when some more testing and commenting is done - should be heavily commented!
It would be perfect if you could do some testing. First whether the fix works and second if there are any side-effects. I will do some testing too - perhaps this evening.

AndKe, 2016-04-30:
Fantastic !  - it works great.
hope this fix gets into master soon.
I must have opened 20-30 logs of different type and size whithout a hiccup , can also close them really fast after the graph windows spawns.  so the fix is perfect.

I also found a 100% reproducible segfault crash with a .tlog (proably one I found in the forums)  - but that's unrelated.
thank you very much

Arne-W, 2016-04-30:
Cool - we made it :tada: :confetti_ball: 
Thank you for supporting me and all the testing you have done. :+1: Good team - really productive!!
I will create a PR this weekend.

Regarding the .tlog crash please create a new issue and upload the .tlog - I am sure we will fix that one too :smiley:

AndKe, 2016-04-30:
> Thank you for supporting me and all the testing you have done. [image:

yep. high-fives all around. :)

as for the other, I already did make a new
https://github.com/ArduPilot/apm_planner/issues/941 , as usual - I am good
at creating new issues, that's the solving-skills that are missing  :)

closed 2016-04-30 by billbonney

AndKe, 2016-05-05:
I'd like to mention that I just experienced a apmplanner crash during an flight with current master.
*not* when loading logs, or using it, I just turned my back for a while, and apmplanner2 was gone when I looked at laptop again. 

there was a segfault at libQt5XcbQpa.so.5.5.1 

in light of the apparent solution to https://github.com/ArduPilot/apm_planner/issues/941  (multithreading related)  - I suspect that the https://github.com/ArduPilot/apm_planner/commit/80a456a1c8d1ccafd5c34ba9dc1946de0b717ec9 fix may need to be applied to other parts of code as well.

billbonney, 2016-05-06:
That's in X, so seems unrelated.  bit more like https://gist.github.com/slimsag/a26d838ccc4480ce21bc for 14.04 (random guess)

Do you have a crash dump or something to glean some clues from?

AndKe, 2016-05-06:
I have no crashdump as I was in a hurry staring the GCS again.
Once https://github.com/ArduPilot/apm_planner/issues/941 is fixed (seems soon)  I'll do some big simulated runs using mavplayback on some of the longest missions  - if there is a crash then I'll report it as a new issue.

commit 80a456a1 (2016-04-30): Graph View: Inserted Semaphore to block parsing thread. Fixes #940

The problem seems to be the termination of the parsing thread before the threadDone() method finishes.
Looks like its related to Qt and/or Ubuntu 16.04