ArduPilot/apm_planner #945: Can't calibrate accelerometer

yazinsai, 2016-05-17:
When I try to run the calibration, I get the below screen:

[image]

Here's the debug file:
```
DEBUG 2016-05-17T22:38:04.083 UAS::executeCommand 241 conf 0 param1 0 param2 0 param3 0 param4 0 param5 1 param6 0 param7 0 
 INFO 2016-05-17T22:38:04.203 STATUS TEXT: 6 : "Initialising APM" 
DEBUG 2016-05-17T22:38:04.204 APM: Text Message rx'd "Initialising APM" 
DEBUG 2016-05-17T22:38:06.220 Param: "INS_PRODUCT_ID" :  QVariant(int, 5) 
DEBUG 2016-05-17T22:38:06.221 Param: "INS_ACCSCAL_X" :  QVariant(double, 1) 
DEBUG 2016-05-17T22:38:06.222 Param: "INS_ACCSCAL_Y" :  QVariant(double, 1) 
DEBUG 2016-05-17T22:38:06.224 Param: "INS_ACCSCAL_Z" :  QVariant(double, 1) 
DEBUG 2016-05-17T22:38:06.225 Param: "INS_ACCOFFS_X" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.226 Param: "INS_ACCOFFS_Y" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.227 Param: "INS_ACCOFFS_Z" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.227 Param: "INS_GYROFFS_X" :  QVariant(double, 0.00276027) 
DEBUG 2016-05-17T22:38:06.228 Param: "INS_GYROFFS_Y" :  QVariant(double, 0.0208213) 
DEBUG 2016-05-17T22:38:06.229 Param: "INS_GYROFFS_Z" :  QVariant(double, -0.00978289) 
DEBUG 2016-05-17T22:38:06.230 Param: "INS_ACC2SCAL_X" :  QVariant(double, 1) 
DEBUG 2016-05-17T22:38:06.230 Param: "INS_ACC2SCAL_Y" :  QVariant(double, 1) 
DEBUG 2016-05-17T22:38:06.231 Param: "INS_ACC2SCAL_Z" :  QVariant(double, 1) 
DEBUG 2016-05-17T22:38:06.238 Param: "INS_ACC2OFFS_X" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.239 Param: "INS_ACC2OFFS_Y" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.239 Param: "INS_ACC2OFFS_Z" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.240 Param: "INS_GYR2OFFS_X" :  QVariant(double, -0.0271287) 
DEBUG 2016-05-17T22:38:06.241 Param: "INS_GYR2OFFS_Y" :  QVariant(double, 0.0292677) 
DEBUG 2016-05-17T22:38:06.241 Param: "INS_GYR2OFFS_Z" :  QVariant(double, -0.00599059) 
DEBUG 2016-05-17T22:38:06.242 Param: "INS_ACC3SCAL_X" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.242 Param: "INS_ACC3SCAL_Y" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.243 Param: "INS_ACC3SCAL_Z" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.243 Param: "INS_ACC3OFFS_X" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.244 Param: "INS_ACC3OFFS_Y" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.244 Param: "INS_ACC3OFFS_Z" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.245 Param: "INS_GYR3OFFS_X" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.245 Param: "INS_GYR3OFFS_Y" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.246 Param: "INS_GYR3OFFS_Z" :  QVariant(double, 0) 
DEBUG 2016-05-17T22:38:06.247 APM: Text Message rx'd "SUCCESS: Executed CMD: 241" 
DEBUG 2016-05-17T22:38:06.247 Rejecting message, check mismatch: current_state:  0  ==  3 , system id  0  ==  1 , comp id  1  ==  1 
```

mboland, 2016-05-17:
I have had this happen at random times with different controllers.
I downloaded the latest version which seemed to fix things.

yazinsai, 2016-05-18:
@mboland thanks for the quick response -- latest version of APM planner, or the beta firmware?

billbonney, 2016-05-18:
[image]
You may need to make sure this option is unchecked

yazinsai, 2016-05-18:
It's unchecked:

[image]

danybony, 2016-07-14:
I'm having the same issue with APM 2.8 and APM Planner v2.0.23-rc1

billbonney, 2016-07-14:
Support requests are handled here http://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0

andrenventer, 2016-07-29:
Guys, lets fix this by making "Text Severity Compatibility" a default "TRUE"

billbonney, 2016-07-29:
@andrenventer that's only correct if you are not running AC3.4+. The fix is to add detection of the APM FW versions and automatically select the correct remapping.

andrenventer, 2016-07-29:
I just installed a fresh apm_planner on mac - updated a PixHawk with latest stable and the Calibration feature for Accel was broken

yazinsai, 2016-07-29:
I ultimately managed to get it working using the CLI (`mavproxy.py`) which worked fine

FlyingCatAlex, 2016-08-04:
In config/tuning tab, check Enable Ardupilot Text severity Compatibility Mode. Then reopen APM software. This works for me.

billbonney, 2016-08-05:
ok, I've add a fix that will auto detect the APM version and do the correct things without doing the checkbox for AC3.4+. Please verify the build here (OSX only at the moment) or build from master.

http://firmware.ardupilot.org/Tools/APMPlanner/daily/2016-08-05/

closed 2016-08-09 by billbonney