ArduPilot/apm_planner #948: Can't build APMPLANNER2 on Ubuntu 16.04

mathojojo, 2016-05-20:
I followed the guide to build apmplanner2.

But it finally failed on make command : 

```
src/uas/UASManager.cc: In member function ‘bool UASManager::setHomePosition(double, double, double)’:
src/uas/UASManager.cc:76:19: error: ‘isnan’ was not declared in this scope
     if (!isnan(lat) && !isnan(lon) && !isnan(alt)
                   ^
src/uas/UASManager.cc:76:19: note: suggested alternative:
In file included from /usr/include/c++/5/random:38:0,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:85,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qalgorithms.h:37,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qlist.h:37,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QList:1,
                 from src/uas/UASManager.cc:11:
/usr/include/c++/5/cmath:641:5: note:   ‘std::isnan’
     isnan(_Tp __x)
     ^
src/uas/UASManager.cc:77:22: error: ‘isinf’ was not declared in this scope
         && !isinf(lat) && !isinf(lon) && !isinf(alt)
                      ^
src/uas/UASManager.cc:77:22: note: suggested alternative:
In file included from /usr/include/c++/5/random:38:0,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:85,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qalgorithms.h:37,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qlist.h:37,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QList:1,
                 from src/uas/UASManager.cc:11:
/usr/include/c++/5/cmath:621:5: note:   ‘std::isinf’
     isinf(_Tp __x)
     ^
Makefile:32793: recipe for target 'build-release/obj/UASManager.o' failed
make: *** [build-release/obj/UASManager.o] Error 1
```

What can I do ? It would be so nice to have a working deb package, instead of building it with many dependencies issues...

Thank you for your help.

dcarpy, 2016-05-23:
What version of Qt?

AndKe, 2016-05-30:
There are some .deb packages here: http://discuss.ardupilot.org/t/download-latest-apm-planner-2-0-20-and-other-releases/8268

For what it's worth - it builds fine on both clean-installed and upgraded 16.04 64bit,  
do a qmake -v   , that will give you answer to the question from dcarpy.

billbonney, 2016-06-09:
fixed with PR #959

closed 2016-06-09 by billbonney