ArduPilot/apm_planner #955: Graph not showing mode arrows correctly

billbonney, 2016-06-04:
This is part of the fix 82ee43d

No the Graph Needs fixing. Table is OK.

Arne-W, 2016-06-04:
Yes - no problem - maybe tomorrow I am a little bit busy today....

Arne-W, 2016-06-05:
@billbonney  Do you have a 3.4 log? It would be great to test the change....

Arne-W, 2016-06-08:
@billbonney Still need a log.....

billbonney, 2016-06-08:
use this one http://discuss.ardupilot.org/t/battery-fail-safe-triggered-at-wrong-voltage/9438/3?u=billb

Arne-W, 2016-06-09:
This is no bug - every thing works like expected :smiley: 
There are just some new Event codes in the ardupilot code. Also the mode got a new data field.
I will update the message formatters to reflect these changes.

billbonney, 2016-06-09:
MAVLink 2 ??? Haven't even thought about handling that

Arne-W, 2016-06-10:
While implementing the the changes I found that the mode messages have changed completely. The first field is now a String containing the name of the mode, the second field holds the mode ID and the third - the new one - holds the reason ID. It's a mess - I do not understand the mixing of  strings and IDs. Either we should use only IDs or strings but mixing is not a good Idea.
Using strings makes it easier cause we don't have to maintain the formatters but using both is the worst solution. @billbonney Do you know if there is a plan to migrate all of those messages to strings or is it only a random change? I think there should be a rule cause otherwise every change in APM code will raise compatibility issues.

billbonney, 2016-06-10:
I don't know, this is related to bin/log files, which are self describing, so we should be able yo handle it fluidly. I haven't looked at the details so I don't know. I can try and find sometime later today

This is mode for AC3.3.3 `FMT, 170, 13, MODE, QMB, TimeUS,Mode,ModeNum`
and for 3.4 it's ``FMT, 172, 14, MODE, QMBB, TimeUS,Mode,ModeNum,Rsn'

Arne-W, 2016-06-11:
Hmm, the problem seems to be that the FMT message does not match the real data!
`FMT, 172, 14, MODE, QMBB, TimeUS,Mode,ModeNum,Rsn` is the FMT, this is the data `MODE, 15931615, Stabilize, 0, 1`. Type code "M" should be a qint8 type and not a string.

billbonney, 2016-06-11:
The issue is the second line is from the exporter. So the error would be in AP2.

Arne-W, 2016-06-11:
Oh yes you are right... I will check that....

Arne-W, 2016-06-12:
Ok after checking the code I'm sure the strings are not generated by the APM-Planner code. We need an original 3.4 dataflash log and tlog to verify this issue.

billbonney, 2016-06-12:
Hi, it's the same logs as link above. https://github.com/ArduPilot/apm_planner/issues/955#issuecomment-224703098

Arne-W, 2016-06-12:
This is neither a tlog nor is it a dataflash log. It is some sort of exported log - perhaps exported by APM planner I don't know. 
I need the original log.

billbonney, 2016-06-12:
mmm. strange. I don't know... I've requested the log fie again, so we can test as it has been removed from the site.

Arne-W, 2016-06-13:
I have found some [other 3.4 dataflash logs](http://discuss.ardupilot.org/t/copter-3-4rc1-rtl-to-rally-point-issue/9659), and they are working fine! This are FMT and Data in the new logs: `FMT, 172, 14, MODE, QMBB, TimeUS,Mode,ModeNum,Rsn` and `MODE, 154950772, 5, 5, 1`. Everything is OK.
I don't know where [this](https://github.com/ArduPilot/apm_planner/issues/955#issuecomment-224703098) exported log comes from - as it is definitly **NOT** an apm-planner export, not compatible and has a corrupt header.
I will create a PR this evening with the updated message formatters and an extended mode message handling the new "reason" field.

commit 9713647c (2016-06-16): Graph View: Adapted Mode message to handle reason and added reason to formatter. Fixes #955
closed 2016-06-16 by billbonney