ArduPilot/apm_planner #957: Error when compiling from source with latest git

martinjuju, 2016-06-07:
System Ubuntu 15.10

When I try to compile from master then I have an error message see below. If I compile from 2.0.21, it works fine.

here is the error message

In file included from src/comm/MAVLinkDecoder.h:38:0,
                 from src/comm/TLogReplayLink.h:5,
                 from src/ui/QGCMAVLinkLogPlayer.h:7,
                 from src/ui/QGCToolBar.h:35,
                 from src/ui/MainWindow.h:74,
                 from src/QGCCore.h:38,
                 from src/QGCCore.cc:32:
src/QGCCore.cc: In member function ‘void QGCCore::initialize()’:
QsLog/QsLog.h:156:11: error: ‘qInfo’ was not declared in this scope
     qInfo()
           ^
src/QGCCore.cc:89:5: note: in expansion of macro ‘QLOG_INFO’
     QLOG_INFO() << "QGCCore::initialize()";
     ^
src/QGCCore.cc: In member function ‘void QGCCore::startLinkManager()’:
QsLog/QsLog.h:156:11: error: ‘qInfo’ was not declared in this scope
     qInfo()
           ^
src/QGCCore.cc:233:5: note: in expansion of macro ‘QLOG_INFO’
     QLOG_INFO() << "Start Link Manager";
     ^
src/QGCCore.cc: In member function ‘void QGCCore::startUASManager()’:
QsLog/QsLog.h:156:11: error: ‘qInfo’ was not declared in this scope
     qInfo()
           ^
src/QGCCore.cc:243:5: note: in expansion of macro ‘QLOG_INFO’
     QLOG_INFO() << "Start UAS Manager";
     ^
Makefile:32625: recipe for target 'build-release/obj/QGCCore.o' failed
make: *** [build-release/obj/QGCCore.o] Error 1

billbonney, 2016-06-07:
You need 5.5.1 for qInfo to compile.  There is a PR to remove this dependency.

martinjuju, 2016-06-07:
Thanks for the info. I will upgrade ubuntu and qt version.

closed 2016-06-07 by martinjuju