ArduPilot/apm_planner #958: Failed to build on Fedora20

night-ghost, 2016-06-08:
I followed the guide to build apmplanner2 but build fails with error:

rc/QGCCore.cc: In member function 'void QGCCore::initialize()':
src/QGCCore.cc:89:11: error: 'qInfo' was not declared in this scope
     QLOG_INFO() << "QGCCore::initialize()";
           ^
src/QGCCore.cc: In member function 'void QGCCore::startLinkManager()':
src/QGCCore.cc:233:11: error: 'qInfo' was not declared in this scope
     QLOG_INFO() << "Start Link Manager";
           ^
src/QGCCore.cc: In member function 'void QGCCore::startUASManager()':
src/QGCCore.cc:243:11: error: 'qInfo' was not declared in this scope
     QLOG_INFO() << "Start UAS Manager";
           ^
make: *** [build-release/obj/QGCCore.o] Error 1
ошибка: Неверный код возврата из /var/tmp/rpm-tmp.vC8yX6 (%build)

Ошибки сборки пакетов:
    Неверный код возврата из /var/tmp/rpm-tmp.vC8yX6 (%build)

Previous version 2.0.16        rc1.32.gc208324 builds OK

billbonney, 2016-06-08:
Duplicate of #957

Arne-W, 2016-06-08:
See PR #959 should fix this. Will build on Qt 5.4.2 again.

billbonney, 2016-06-09:
PR #959 is merged so closing. Tested build with Qt5.4.2 on OSX

closed 2016-06-09 by billbonney