ArduPilot/apm_planner #961: Dependency problems on Ubuntu 14.04 - cannot install

pwbecker, 2016-06-13:
When trying to install the latest deb I get errors as follows:

sudo dpkg -i apm_planner_2.0.20_ubuntu_trusty64.deb 
Selecting previously unselected package apmplanner2.
(Reading database ... 420984 files and directories currently installed.)
Preparing to unpack apm_planner_2.0.20_ubuntu_trusty64.deb ...
Unpacking apmplanner2 (2.0.20) ...
dpkg: dependency problems prevent configuration of apmplanner2:
 apmplanner2 depends on libflite1 (>= 1.4-release-9~); however:
  Version of libflite1:amd64 on system is 1.4-release-8ubuntu0.1.
 apmplanner2 depends on libqt5core5a (>= 5.4.0); however:
  Version of libqt5core5a:amd64 on system is 5.2.1+dfsg-1ubuntu14.3.
 apmplanner2 depends on libqt5gui5 (>= 5.3.0) | libqt5gui5-gles (>= 5.3.0); however:
  Version of libqt5gui5:amd64 on system is 5.2.1+dfsg-1ubuntu14.3.
  Package libqt5gui5-gles is not installed.
 apmplanner2 depends on libstdc++6 (>= 5.2); however:
  Version of libstdc++6:amd64 on system is 4.8.4-2ubuntu1~14.04.3.

dpkg: error processing package apmplanner2 (--install):
 dependency problems - leaving unconfigured

HiddenMarkov, 2016-06-13:
I am having the same problem. I was able to resolve the libflite issue by manually installing the correct version, but I am unable to do the same for the libqt5* or libstdc++ libraries.

pwbecker, 2016-06-13:
I see that in DEBIAN/control within the .deb file is the line below.

Perhaps this just needs to be edited to allow lower versions?  Or does the code depend on features of those latest versions?

Depends: libasound2 (>= 1.0.16), libc6 (>= 2.14), libflite1 (>= 1.4-release-9~), libgcc1 (>= 1:4.1.1), libqt5core5a (>= 5.4.0), libqt5gui5 (>= 5.3.0) | libqt5gui5-gles (>= 5.3.0), libqt5network5 (>= 5.0.2), libqt5opengl5 (>= 5.0.2) | libqt5opengl5-gles (>= 5.0.2), libqt5printsupport5 (>= 5.0.2), libqt5qml5 (>= 5.0.2), libqt5quick5 (>= 5.0.2) | libqt5quick5-gles (>= 5.0.2), libqt5script5 (>= 5.0.2), libqt5serialport5 (>= 5.1.0), libqt5sql5 (>= 5.0.2), libqt5svg5, libqt5test5 (>= 5.0.2), libqt5widgets5 (>= 5.2.0), libsdl2-2.0-0 (>= 2.0.0), libsndfile1 (>= 1.0.20), libssl1.0.0 (>= 1.0.0), libstdc++6 (>= 5.2), zlib1g (>= 1:1.1.4), qtdeclarative5-qtquick2-plugin | qml-module-qtquick2

Arne-W, 2016-06-14:
As I have build and run the actual master on a 14.04, 15.10 and 16.04 ubuntu machine it seems to be a bug in the dependency. @pwbecker As you said it should work if you change those dependencies. Unfortunately I never created a .deb installer so I cannot really help you.

Arne-W, 2016-06-14:
perhaps you want to try [this](https://github.com/ArduPilot/apm_planner/issues/963#issuecomment-225908992)

pwbecker, 2016-06-14:
Ok, some progress... I edited the control file and rebuilt the deb, uninstalled the old apmplanner2, installed the new one, and there were no dependency issues. So far so good...

However, when trying to run apmplanner2, I get

```
apmplanner2: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by apmplanner2)
apmplanner2: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by apmplanner2)
```

If i try `# apt-get install libstdc++6`
then i get: `libstdc++6 is already the newest version`

Arne-W, 2016-06-14:
Have you seen my comment above? There is a link where you can get an apm-planner built on 14.04 - but you will need Qt 5.5.1 
That's the only thing I can do at the moment.

pwbecker, 2016-06-14:
I did see it, but was looking for another way.  I find it adds admin when upgrading to have some software installed by unzipping into a folder, and others via apt-get.

So far however it looks like apmplanner2 is compiled depending on shared libraries more recent than can be installed on Ubuntu 14.04 :(

Arne-W, 2016-06-14:
Ok - the link should provide only a temporary solution - better than having no planner at all :smiley:
@billbonney could you please check if there is something we can do to make the deb package work on 14.04?

dcarpy, 2016-06-15:
@pwbecker Try apm_planner_2.0.23-rc1_ubuntu_trusty64.deb at https://www.dropbox.com/sh/6tx02zn95yktucy/AABaCuoi1a2_GjUNALeWMoKia?dl=0
It was built on 14.04 using Qt 5.4.2 static libs.

HiddenMarkov, 2016-06-15:
@dcarpy 

Thank you for this, this seems to have gotten me much closer, but upon launching APM I get this error:

`Unable to load ApmToolbar.qml. Please reinstall the application and try again. Errors are as follows:
file:///usr/share/APMPlanner2/qml/ApmToolBar.qml:17:1: module "QtQuick" is not installed`

Line 17 of ApmToolBar.qml:

`import QtQuick 2.0`

I've double checked that I have the QtQuick plugin installed using

`dpkg -s qtdeclarative5-qtquick2-plugin`

With the resulting output:

`Package: qtdeclarative5-qtquick2-plugin
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 295
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Multi-Arch: same
Source: qtdeclarative-opensource-src
Version: 5.2.1-3ubuntu15.1`

So it appears that I do indeed have QtQuick installed. I've tried re-installing but I get the same result. Any Ideas? I am guessing there is some broken env variable or something, but I am fairly mediocre at linux... (I am using ubuntu 14.04 because I need to run ROS indigo)

dcarpy, 2016-06-15:
@HiddenMarkov Hmm...interesting.  I would have thought the apmplanner2 binary contained all the static libs it needed.  Try this:
Install PPA:
sudo apt-add-repository ppa:beineri/opt-qt542-trusty
sudo apt-get update
sudo apt-get install qtdeclarative5-qtquick2-plugin
More info about the PPA:
https://launchpad.net/~beineri/+archive/ubuntu/opt-qt542-trusty

HiddenMarkov, 2016-06-16:
@dcarpy

 Thank you for the idea but unfortunately doing that does not seem to resolve the issue. If i go into Qt Creator and go to 'help->about plugins' QtQuick does show up as installed there as well.

dcarpy, 2016-06-16:
Dang.  Maybe install libqt5quick5?
sudo apt-get install libqt5quick5

dcarpy, 2016-06-16:
Or install qml-module-qtquick2.  Looking closer on Qt 5.4.2, qtdeclarative5-qtquick2-plugin is a transitional dummy package.

pwbecker, 2016-06-16:
I have the same issue as Mr Markov.  I have the following installed:
```
ii  qtdeclarative5-qtquick2-plugin:amd64                        5.2.1-3ubuntu15.1                                   amd64        Qt 5 Qt Quick 2 QML plugin
ii  libqt5quick5:amd64                                          5.2.1-3ubuntu15.1                                   amd64        Qt 5 Quick library
ii  libqt5quickparticles5:amd64                                 5.2.1-3ubuntu15.1                                   amd64        Qt 5 Quick particules module
ii  libqt5quicktest5:amd64                                      5.2.1-3ubuntu15.1                                   amd64        Qt 5 Quick Test library

```
Should the .deb file not specify its dependencies?

pwbecker, 2016-06-16:
So I looked to install qml-module-qtquick2:
```
sudo apt-get -s install qml-module-qtquick2
[sudo] password for peterb: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package qml-module-qtquick2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
E: Package 'qml-module-qtquick2' has no installation candidate
```

HiddenMarkov, 2016-06-17:
I believe qml-module-qtquick2 is only available on wily and higher. http://packages.ubuntu.com/search?keywords=qml 

I'm going to try building myself and see if that works out.

pwbecker, 2016-06-18:
Would like to hear how that goes @HiddenMarkov 

At the moment for me Amplanner2 still runs, but without the toolbar, although the View menu can be used to access the functionality.

Seems like the developers added a new library for an updated toolbar without looking too closely at availability for LTS versions :(

billbonney, 2016-06-20:
@pwbecker You need to install Qt5.4.2 or greater on your 14.04 install for the bid to work. This is as simple as adding a PPA for Qt.5.4.2 or greater. Qt5.5.1 can also be used. e.g.. 

https://launchpad.net/~beineri/+archive/ubuntu/opt-qt551-trusty

Let's keep support requests to http://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0 and posts findings there.

Thx.

@HiddenMarkov You confusing availability of Qt being linked to a Ubuntu release. It is not. Versions will run on almost all from 12.04. The question is what is default version shipped with Ubuntu and it's always a much older version that is current. You should be able to build APM Planner on any version of ubuntu. I would suggest Qt5.5.1 as the preferred starting version.

HiddenMarkov, 2016-06-22:
@billbonney 
@pwbecker 
@dcarpy 

By request of billbonney, I have created a new troubleshooting topic here:
http://discuss.ardupilot.org/t/problems-installing-and-or-building-2-0-20-on-ubuntu-14-04/9819

pwbecker, 2016-06-22:
Thanks for that, I will look there.  Just as a closing comment, deployment problems are IMHO a development issue, and not a support issue, although in my experience dev will usually try and offload it onto support.  The question above w.r.t. incorrect dependencies in the .deb appear to be unanswered so far...

karikalansaitechnology, 2017-05-09:
why i cannot install dpkg-dev package i mentioned below this
E: Package 'dpkg-dev' has no installation candidate

carlosnewmusic, 2018-05-06:
libssl1.0.0 (>= 1.0.0) Unresolved dependency, when I have an upper version installed libssl1.0.2

billbonney, 2018-05-14:
@carlosnewmusic You would have more success in building the code following the instructions here https://github.com/ArduPilot/apm_planner/blob/master/README.md

It really isn't that hard, the thing is it means it relaxes the version of the packages required when you build from source.

I am not sure which version of Qt you are using but Qt5.6.2 to Qt 5.9.3 should work ok

carlosnewmusic, 2018-05-14:
@billbonney compilation terminated.
Makefile:5911: fallo en las instrucciones para el objetivo 'build-release/obj/gpsitem.o'
make: *** [build-release/obj/gpsitem.o] Error 1

billbonney, 2018-05-15:
@carlosnewmusic I added instructions to the discussion forum see https://discuss.ardupilot.org/t/problems-installing-and-or-building-2-0-20-on-ubuntu-14-04/9819/23

Arne-W, 2023-07-15:
Very outdated....

closed 2023-07-15 by Arne-W