ArduPilot/apm_planner #963: ERROR 2016-06-14T12:28:59.559 bind failed! QHostAddress( "0.0.0.0" ) : 14550

matsta, 2016-06-14:
Hi,

Unable to flash install new firmware on my Pixhawk.

Open apmplanner and choose my criterias, frame etc. The /dev/ttyAC* connection is showing and my user on the comp has both RW and is a memeber of the dialout group.

Clicking OK on the confirm dialog and the apmplanner output pane logs:
Started downloading http://firmware.diydrones.com/Copter...uCopter-v2.px4
Finished downloading /tmp/APM Planner.Sa7964
Requesting Board ID
Requesting Board Rev
Requesting FW Size
Requesting CoA
Requesting board Serial Number
COA Verification Success
Erasing board <===== Here it hangs.

I'm unable to go pass this stage.

Looking in the log and I can see some errors being logged:
ERROR 2016-06-14T12:28:59.559 bind failed! QHostAddress( "0.0.0.0" ) : 14550
DEBUG 2016-06-14T12:28:59.597 SCW: Link is disconnected
DEBUG 2016-06-14T12:28:59.602 Baud rate: 115200 Expected: 4
DEBUG 2016-06-14T12:29:04.499 "retrieve versionobject from server: http://firmware.diydrones.com/Tools/APMPlanner/apm_planner_version.json"
DEBUG 2016-06-14T12:29:04.767 Downloading: 342 / 342
DEBUG 2016-06-14T12:29:04.768 Downloading: 342 / 342
DEBUG 2016-06-14T12:29:04.769 AutoUpdateCheck::httpFinished()
ERROR 2016-06-14T12:29:04.772 Error evaluating version object
ERROR 2016-06-14T12:29:04.773 Error @line# 1
ERROR 2016-06-14T12:29:04.774 Backtrace: ()
ERROR 2016-06-14T12:29:04.774 Syntax Check: ""
ERROR 2016-06-14T12:29:04.775 Syntax Check line: 1 col: 1
DEBUG 2016-06-14T12:29:12.878 Droneshare: ok notice until next update
DEBUG 2016-06-14T12:29:31.796 flashButtonClicked
DEBUG 2016-06-14T12:29:31.797 Flash button clicked without any HTTP links!

These two looks strange:
Baud rate: 115200 Expected: 4
bind failed! QHostAddress( "0.0.0.0" ) : 14550

I'm on ubuntu 14.04 64-bit. Apmplanner version 2.0.17, have same problem with 2.0.18. haven't tried 2.0.20 since it is incompatible with 14.04 due to dependencies.

Arne-W, 2016-06-14:
You must use at least 2.0.21. There was an update to the URLs and the old APM-planner does not support redirection.

matsta, 2016-06-14:
Is there any guide on how to install 2.0.21 on ubuntu 14.04 LTS? The 2.0.20
guide is insufficient an does not work due to dependency problems. Any PPA?

Can you supply me with an URL to the working version?

BR

Arne-W, 2016-06-14:
try this - build on ubuntu 14.04 with Qt 5.5.1
You will need to update your Qt version Ubuntu 14.04 brings 5.2.1
[release.zip](https://github.com/ArduPilot/apm_planner/files/314229/release.zip)

matsta, 2016-06-14:
Thanks Arne, that worked.

Only problem now is that Accel Calib fails du to an Error: CMD 241
 here is the DEBUG log:
[20160614 19:47:29.949 DEBUG] - Severity: 3  text: "PreArm: Need 3D Fix"
[20160614 19:47:47.447 DEBUG] - APMToolBar: INITIAL SETUP SELECTED
[20160614 19:47:47.448 DEBUG] - APMToolBar: clear selected buttons
[20160614 19:47:47.448 DEBUG] - APMToolBar: FLIGHT DATA unselected
[20160614 19:47:47.448 DEBUG] - APMToolBar: INITIAL SETUP unselected
[20160614 19:47:47.557 DEBUG] - APMToolBar: INITIAL SETUP SELECTED
[20160614 19:47:57.371 DEBUG] - UAS::executeCommand 241 conf 0 param1 0
param2 0 param3 0 param4 0 param5 1 param6 0 param7 0
[20160614 19:47:57.386 INFO ] - STATUS TEXT: 3 : "Place vehicle level and
press any key.\r"
[20160614 19:47:57.386 DEBUG] - APM: Text Message rx'd "Place vehicle level
and press any key.\r"
[20160614 19:47:57.386 DEBUG] - Severity: 3  text: "Place vehicle level and
press any key.\r"
[20160614 19:47:57.389 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:47:57.389 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:47:57.389 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:02.391 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:48:02.392 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:48:02.392 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:07.390 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:48:07.390 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:48:07.391 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:12.390 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:48:12.391 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:48:12.402 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:17.392 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:48:17.393 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:48:17.393 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:22.403 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:48:22.403 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:48:22.404 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:27.402 INFO ] - STATUS TEXT: 1 : "Initialising APM..."
[20160614 19:48:27.402 DEBUG] - APM: Text Message rx'd "Initialising APM..."
[20160614 19:48:27.403 DEBUG] - Severity: 1  text: "Initialising APM..."
[20160614 19:48:27.404 INFO ] - STATUS TEXT: 3 : "Calibration FAILED\r"
[20160614 19:48:27.404 DEBUG] - APM: Text Message rx'd "Calibration
FAILED\r"
[20160614 19:48:27.404 DEBUG] - Severity: 3  text: "Calibration FAILED\r"
[20160614 19:48:27.405 DEBUG] - APM: Text Message rx'd "FAILURE: Failed
CMD: 241"
[20160614 19:48:27.405 DEBUG] - Severity: 0  text: "FAILURE: Failed CMD:
241"
[20160614 19:48:27.477 DEBUG] - Rejecting message, check mismatch:
current_state:  0  ==  3 , system id  0  ==  1 , comp id  1  ==  1

Any idea on why I'm getting this error?

For anyone else that needs to update the qt packages on Ubuntu trusty 14.04:
More info here:
https://launchpad.net/~beineri/+archive/ubuntu/opt-qt551-trusty
Install PPA:
sudo apt-add-repository ppa:beineri/opt-qt551-trusty
sudo apt-get update
sudo apt-get install qt-latest
source /opt/qt55/bin/qt55-env.sh

Worked for me.

Arne-W, 2016-06-14:
Hmmm - I don't know. Just an Idea - have you checked the "Enable Ardupilot Text Severity Compatibilty Mode (pre-AC3.4)"? You can find it the Config/Tuning menue.

matsta, 2016-06-14:
Should it be checked or not set? Is that a backward compatible value or
what will it do when set?

matsta, 2016-06-15:
That didn't work. But I reverted to version 2.0.18 after I got qt updated
and was able to complete the accel calibration.

However for 2.0.18: clicking on config/tuning and then trying to open
Standard Params yields a black screen. No parameters gets loaded. This
works in the version you posted.

So there is something broken between 2.0.18 and 2.0.21(if that's the
version you posted).

Any ideas?

Arne-W, 2016-06-15:
Which ArduPilot version are you using?

matsta, 2016-06-15:
Hi,

How can I check that? I'm on the version downloaded by the zipped 2.0.21
version you posted here. X-style frame.

matsta, 2016-06-15:
Pixhawk, if that is what you mean.

Arne-W, 2016-06-15:
No I mean the software version which is flashed on you Pixhawk!

matsta, 2016-06-15:
How can I determine the version?

Arne-W, 2016-06-15:
Just load a dataflash log, the mode messages should show something like this:
[image]
This example shows APM:Copter 3.4-rc1

matsta, 2016-06-15:
Don't have any flash logs, unfortunately I removed the old installation
path. But look at the attached image and I think the version is shown
there. 3.3.3 seems to be the version I flashed with yesterday. And my unit
seems to be PX4 FMU 2.x

matsta, 2016-06-15:
From the console log:
[20160615 9:59:05.965 DEBUG] - Inserting  "ttyACM0"
[20160615 9:59:05.966 DEBUG] - Inserting  "rfcomm0"
[20160615 9:59:05.969 DEBUG] - Detected:  "PX4 FMU v2.x"  : 17

matsta, 2016-06-15:
looks like my attached image didn't go through. Here is an imgur link:
http://imgur.com/uAndmcp

Arne-W, 2016-06-15:
All right thanks for the info. I will try to reproduce this issue in the next days and perhaps fix it. So stay tuned I will write as soon as I have new infos.

matsta, 2016-06-15:
OK, just to sum up, bugs I found so far with the version you posted:

Both with and without  "Enable Ardupilot Text Severity Compatibilty Mode
(pre-AC3.4)":
Calibrating the compass fails, seems fine when calibrating but then prearm
check gives "inconsistent compassus". This woks if I use version 2.0.18.
However disconnect the connection with the pixhawk from within apmplanner,
unplug and plug back in your Pixhawk, restart apmplanner, and reconnect
with the pixhawk, prearm check and compass is calibrated. So that works.

Accel Calibration doesn't work at all, fails with error: CMD 241. Works
with version 2.0.18.

billbonney, 2016-06-16:
You need to restart APM Planner 2.0 between changing the AC3.4 compatibly setting. I just tested with AC3.3.3 and it works.

Support request are discussed here http://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0

Closing this thread as github is not a support forum. Thx for your understanding.

closed 2016-06-16 by billbonney