ArduPilot/apm_planner #966: Debian package

mikolajzieba, 2016-06-29:
Hi, 
Is there any possibility to deploy Debian package for APM?
For Debian jessie there is problem with two dependencies:
 apmplanner2 depends on libqt5core5a (>= 5.4.0); however:
 In repo libqt5core5a 5.3.2+dfsg-4+deb8u1.
 apmplanner2 depends on libstdc++6 (>= 5.2); however:
 In repo libstdc++6:amd64  4.9.2-10.

mikolajzieba, 2016-07-04:
I have found a solution. 
For all who want to use apm_planner on Debian Jessie: https://github.com/mikolajzieba/apm_planner_debian_jessie/

billbonney, 2016-07-04:
You can always build Qt from source for any platform it supports (it actually fairly straight forward)

https://wiki.qt.io/Building_Qt_5_from_Git

You can make it work with Qt5.2 or Qt5.3. There is one dependency in there, but for our main builds we are Qt5.4+ just for simplicity. (I build with Qt 5.5.1 or 5.6.1 on OSX)

closed 2016-07-04 by billbonney