ArduPilot/apm_planner #969: Can't build apm_planner on OSX 10.11.5

zhydanovajulia, 2016-07-08:
While doing `make -j8`:

```
/usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qpair.h:62:44: error: /usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qstring.h:/usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qstring.h756::75655::55 : error: error: /usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qpair.h:62:44: error: /usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qpair.h:62:44: error: /usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qstring.h:756:55: error: /usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qpair.h:62:44: error: /usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers/qpair.h:62:44: error: no member named 'declval' in namespace 'std'
no notype  namedtype  'u16string'named  in'u16string'  namespacein 'std'
 namespace 'std'
no member named 'declval' in namespace 'std'
no member named 'declval' in namespace 'std'
no type named 'u16string' in namespace 'std'
no member named 'declval' in namespace 'std'
no member named 'declval' in namespace 'std'
    static inline QString fromStdU16String(const std::u16string &s);        Q_DECL_NOEXCEPT_EXPR(noexcept(std::declval<T1&>() = p.first) && noexcept(std::declval<T2&>() = p.second))        Q_DECL_NOEXCEPT_EXPR(noexcept(std::declval<T1&>() = p.first) && noexcept(std::declval<T2&>() = p.second))
```
```
fatal error: too many errors emitted, stopping now [-ferror-limit=]
15 errors generated.
make: *** [build-release/obj/diagnostics.o] Error 1
make: *** Waiting for unfinished jobs....
20 errors generated.
20 errors generated.
make: *** [build-release/obj/languagetype.o] Error 1
make: *** [build-release/obj/cacheitemqueue.o] Error 1
20 errors generated.
make: *** [build-release/obj/alllayersoftype.o] Error 1
20 errors generated.
20 errors generated.
make: *** [build-release/obj/kibertilecache.o] Error 1
make: *** [build-release/obj/memorycache.o] Error 1
20 errors generated.
make: *** [build-release/obj/cache.o] Error 1
20 errors generated.
make: *** [build-release/obj/opmaps.o] Error 1
```

and so on. 
```
 qmake -v
QMake version 3.0
Using Qt version 5.6.1 in /usr/local/Cellar/qt5/5.6.1-1/lib
```
Please help.

billbonney, 2016-07-08:
We do support here http://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0

Please raise it there

AP2 will build with Qt5.6 which means you have a problem with your Dev env.

closed 2016-07-08 by billbonney

zhydanovajulia, 2016-07-08:
ok, thanks