ArduPilot/apm_planner #971: Errors in Linux installation documention

zlite, 2016-07-29:
I have sucessfully installed APM Planner on Ubuntu running on a Chromebook, and had to solve the following problems that came up from wrong or incomplete documentation:

1) To get access to the telemetry radio, the user must be added to group *serial*, not *dialout*. Thus: "sudo adduser [user] serial". The instruction to use group dialout pops up on first use. This should be changed, and added to the documentation. It's unclear whether user root must also be added to the serial group, but I did it just to be safe. 
2) The program crashes on connect unless "Mute all audio output" is selected

billbonney, 2016-07-29:
Thanks for the info. I suspect that Flite Audio is not supported on the chrome book, or it's Flitev2.

zlite, 2016-08-01:
A few more issues. On [this page](http://ardupilot.org/planner2/docs/installation-for-linux.html), the instructions refer to "apmplanner*.deb" but the file naming has changed. It's now "apm_planner*.deb"

zlite, 2016-08-01:
Also, my installation on Ubuntu 16 fails with the following errors, despite running "sudo apt-get -f install" and trying again
 
(trusty)chris@localhost:~/Downloads$ sudo dpkg -i apm_planner*.deb
[sudo] password for chris: 
Selecting previously unselected package apmplanner2.
(Reading database ... 71653 files and directories currently installed.)
Preparing to unpack apm_planner_2.0.8_debian64.deb ...
Unpacking apmplanner2 (2.0.8) ...
dpkg: dependency problems prevent configuration of apmplanner2:
 apmplanner2 depends on libopenscenegraph80; however:
  Package libopenscenegraph80 is not installed.
 apmplanner2 depends on libudev0 (>= 0.140); however:
  Package libudev0 is not installed.

dpkg: error processing package apmplanner2 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 apmplanner2

zlite, 2016-08-01:
Correction: I'm using Ubuntu 14 (trusty)

billbonney, 2016-08-02:
2.0.8.  Wow we are on 2.0.23-rc1

See discuss.ardupilot.org for links to the new released

zlite, 2016-08-02:
Ah, my bad. The list of downloads is upside down! (earliest first, for some reason. Should be inverse chronological order)

http://firmware.ardupilot.org/Tools/APMPlanner/

billbonney, 2016-08-02:
Cool, I cleaned that up to show only 2.0.20 in the root dir... I will remove that as well soon, since the final solution will have the latest in the latest dir. Thx :)

zlite, 2016-08-02:
Excellent. Good solution. 

2.0.20 installed perfectly!

dcarpy, 2016-08-02:
We moved to Qt 5.4.2 as a minimum. See this thread:
http://discuss.ardupilot.org/t/problems-installing-and-or-building-2-0-20-on-ubuntu-14-04/9819/1

zlite, 2016-08-02:
A thought on version naming. Normally people stay with single digits in each version level: x.x.1 - x.x.9 and then rather than going to the confusing x.x.10, they increment the next level up. So I think 2.0.9 should have been followed by 2.1.0, rather than 2.0.10. This avoids the confusion where 2.0.8 looks newer than 2.0.20

billbonney, 2016-08-02:
> Normally people stay with single digits in each version level: x.x.1 - x.x.9 and then rather than going to the confusing x.x.10

I can show you many places where that is not true. I am using a Laptop running OSX 10.11.6. And I was just using Remote Desktop from Microsoft at 8.0.33.

I think just having the latest stable release in the root without the clutter helps avoid the confusion.

zlite, 2016-08-02:
Fair enough.  I'm just an ordinal pedant ;-)

Arne-W, 2017-07-01:
Seems to be solved....
If nobody complains I will close this issue in the next days.

closed 2020-05-17 by Arne-W

islemhm, 2021-05-09:
hello i have a problem in APMplanner, i use ubuntu 18.04 and this version of apm planner "apm_planner_2.0.26_bionic64.deb"This is my ‘install firmware’ screen which is blank in both disconnected & connected state (I’m trying to configure my pixhawk1 for the first time)
Any ideas why this is occuring and how do I fix this?
[image]

islemhm, 2021-05-10:
Arne Wischmann  Thank you for your help I installed the version and I encountered a problem to  install firmware I load ardurover but when I connect the card there is the copter which appeared
[image]
[image]

Arne-W, 2021-05-12:
Sorry I never tested rover firmware as I only have a Copter. So Sorry.

islemhm, 2021-05-12:
@Arne-W thank you for your help