ArduPilot/apm_planner #972: Multi-UAV Setup Problem

jmodares, 2016-07-30:
I'm doing some UAS swarm simulations using SITL and APM Planner. When I run more than 14 SITL instances and connect them to APM Planner, it starts to print this message:
"MAVLink Baud Rate Mismatch" "Please check if the baud rates of APM Planner and your autopilot are the same."
If I run more than 17 SITL instances, it'll crashes with "segmentation fault". I've tested with QGroundCointrol and it works fine.

billbonney, 2016-07-30:
What version and OS you using?

jmodares, 2016-07-30:
APM Planner v2.0.23-rc1 (I clone it from here)
OS:
1) Linux Mint 17.3
2) Ubuntu 14.04

billbonney, 2016-08-04:
I'm not sure what this issue is.  If you need this fixed as priority, please email me (see profile for address)