ArduPilot/apm_planner #977: OSX 2.0.23RC4 crashes on startup

lvale, 2016-08-21:
Won’t start. Already tested via the upgrade link and with the directly downloaded DMG from firmware.ardupilot.org:

2.0.23RC3 works great

Crash report from the OS

Missing QT libs
System Integrity Protection: enabled
Crashed Thread: 0
Exception Type: EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000002, 0x0000000000000000
Exception Note: EXC_CORPSE_NOTIFY
Application Specific Information:
dyld: launch, loading dependent libraries
Dyld Error Message:
Library not loaded: @rpath/QtOpenGL.framework/Versions/5/QtOpenGL
Referenced from: /Applications/APM Planner 2.0.app/Contents/MacOS/apmplanner2
Reason: image not found
Binary Images:
0x7fff6cb29000 - 0x7fff6cb60a47 dyld (360.22) <DC81CC9D-651A-3A45-8809-928282052BD3> /usr/lib/dyld
0x7fff8b7f5000 - 0x7fff8b82bfff libssl.0.9.8.dylib (59.60.1) <C047F10D-A929-36C8-83DE-E8BB7183A158> /usr/lib/libssl.0.9.8.dylib
0x7fff93a8a000 - 0x7fff93a8afff com.apple.ApplicationServices (48 - 48) <ADD57D3A-142F-3EF5-BFD8-EACD82164884> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices
0x7fff96ecb000 - 0x7fff96edcff7 libz.1.dylib (61.20.1) <B3EBB42F-48E3-3287-9F0D-308E04D407AC> /usr/lib/libz.1.dylib
0x7fff9b627000 - 0x7fff9b70dff7 libcrypto.0.9.8.dylib (59.60.1) <D68067AD-D1E0-3196-9796-51BE9B969C8E> /usr/lib/libcrypto.0.9.8.dylib

billbonney, 2016-08-21:
It seems it didn't copy the frameworks over on this new build. This one is Qt5.6.1 not Qt5.4.2 as previous builds.. I'll post a new version later. Should be straight forward fix

billbonney, 2016-08-21:
If you redownload the binary, its fixed now to include the missing frameworks

see http://firmware.ardupilot.org/Tools/APMPlanner/beta/apm_planner_2.0.23-rc4_osx.dmg

closed 2016-08-21 by billbonney