ArduPilot/apm_planner #978: Memory Leak Issue

jmodares, 2016-08-23:
The APM Planner uses roughly around 100 MiB of memory, and when it connects to a SITL, it starts to use more and more memory over the time,

Qt:
1) 5.5.1

APM Planner:
v2.0.23-rc1

OS:
1) Linux Mint 17.3
2) Ubuntu 14.04

billbonney, 2016-08-23:
How much more and over what time frame. 100MB is not really a lot of memory of machine with memory usually being 2 or 4 GB minimum.

jmodares, 2016-08-23:
The initial memory is not that important event you start with 1 GiB, but the increase rate is important, which is about 1 MiB per minute. It means that you can not run the APM Planner for a whole day

dcarpy, 2016-08-25:
Interesting.  On Linux it's about 5.6MB per minute.  On Windoze about 5MB per minute.
I built on Windows using Qt 5.6.1--didn't help, still leaks. It doesn't leak when not connected, but I guess that makes sense. Well, my valgrind skills are a little rusty...but here we go.

billbonney, 2016-08-25:
I'm sure it the auto graphing feature just make it switch off/on able

billbonney, 2016-08-25:
Here's AP2 running for 1d7hr
[image]
[image]

billbonney, 2016-08-26:
It's still running, even with the memory leak... And it's connected to a simulator flying the whole time. For what it's worth!