ArduPilot/apm_planner #979: Feature Request: Add Solo to firmware upload

jmachuca77, 2016-08-28:
As part of a new initiative to support Solo on master, we need to support a way of uploading custom firmware to Solo. The process is fairly simple, the firmware file should be uploaded into /firmware on Solo, and then the command loadPixhwak.py should be issued in Solo. 
Solo_Cli handles this on the file flash.py function flash_px4. They also erase the contents of the folder /firmware/loaded.

billbonney, 2016-08-28:
Seems simple enough. What is the transfer mechanism `scp` or `ftp`

AndKe, 2016-12-01:
SCP works fine.

jmachuca77, 2018-08-16:
not sure if this was implemented or not, but now Mission planer and QGC both support it as well as other GSCs that are more suited for use with the solo. So closing.

billbonney, 2018-08-16:
This is still a valid request for APM Planner to support SOLO upload, reopening.