ArduPilot/apm_planner #980: Graph: Error execing index: TEC2 UNIQUE constraint failed: INDEX.idx Unable to fetch row

AndKe, 2016-09-09:
Please see attached log.
I were asked to analyze it, (crash) - but instead discovered that APM Planner returns an error.
[12.BIN.zip](https://github.com/ArduPilot/apm_planner/files/463790/12.BIN.zip)

Arne-W, 2016-09-11:
Okay I made a first check. The log is somehow corrupted. As far as I can see the main problem is a FMT message arriving during data parsing. Normally all FMT (fomat) messages are at the beginning of the log, than all parameters and after the the binary Log data. In this log a FMT message is read while the parser is already parsing the binary log data. This leads to an Index error in the database which stores the data.
I will see if can provide some sort of hot-fix.... perhaps today...

AndKe, 2016-09-11:
Great.
It is strange, still, all I can say, is that the user is certainty using
stock,official release (not some modified version)

Arne-W, 2016-09-11:
See my new PR #981 - should fix the problem.

commit 0cc6f92a (2016-09-13): Fixes #980 Datamodel now accepts FMT inserts after data inserts.
closed 2016-09-13 by billbonney