ArduPilot/apm_planner #991: 3DR radio flashing error

AndKe, 2016-11-28:
(terminal output)
```
Flashing with firmware
./sik_uploader/firmware/1.9/radio~hm_trp.ihx
..............................uploading to port /dev/ttyUSB0
Connecting to /dev/ttyUSB0
Traceback (most recent call last):
  File "./sik_uploader/sik_uploader.py", line 278, in <module>
    if not up.check():
  File "./sik_uploader/sik_uploader.py", line 237, in check
    self.autosync()
  File "./sik_uploader/sik_uploader.py", line 197, in autosync
    import fdpexpect, time , pexpect
ImportError: No module named fdpexpect

Complete 1
```
pexpect is installed using pip install..

AndKe, 2017-02-02:
also, when at it, please add option for custom firmware.

RHHoek, 2017-02-22:
I solved this issue (Ubuntu 16:10) by installing these packages:
apt-get install python-serial python-pexpect

and changing sik_uploader/sik_uploader.py:

line 197,198
         import fdpexpect, time , pexpect
         ser = fdpexpect.fdspawn(self.port.fileno(), logfile=sys.stdout)
to:
         import pexpect, time , pexpect.fdpexpect
         ser = pexpect.fdpexpect.fdspawn(self.port.fileno(), logfile=sys.stdout)
line 200
             self.port.setBaudrate(self.atbaudrate)
to:
             self.port.baudrate=self.atbaudrate
line 219
                     self.port.setBaudrate(115200)
to:
                     self.port.baudrate=115200
line221
             except fdpexpect.TIMEOUT:
to:
             except pexpect.fdpexpect.TIMEOUT:
line 224
             self.port.setBaudrate(115200)
to:
             self.port.baudrate=115200

Arne-W, 2017-02-22:
Nice :+1: 
It would be great if you could make a pull request.

RHHoek, 2017-03-02:
I am not ready to send you a pull request at the moment (need to find out in detail how this works). So for the moment I can only provide the changed file.
[sik_uploader.py.zip](https://github.com/ArduPilot/apm_planner/files/814350/sik_uploader.py.zip)

AndKe, 2017-03-02:
@RHHoek , works fine
@Arne-W  here's a PR https://github.com/ArduPilot/apm_planner/pull/1009
- where should I put the dependencies ?

Arne-W, 2017-03-02:
Awesome guys! 
The dependencies - good question - perhaps @billbonney knows :wink:

AndKe, 2017-03-02:
added dependencies as best I could  :)

RHHoek, 2017-03-08:
perhaps you also need to add these dependencies in de README.md ?

AndKe, 2017-03-08:
Good point , done :)

AlivaDot, 2017-03-15:
Hello!
I received this problem too.
But find another solution for OSX (10.12.3)
First of all install some packages:
```
sudo easy_install pip
sudo python -m pip install pyserial
sudo python -m pip install pexpect
```
(pyserial because error was at step "import sys, argparse, binascii, serial, glob")
And some changes in sik_uploader.py line 197
```
- import fdpexpect, time , pexpect
+ import time , pexpect
+ from pexpect import fdpexpect
```

Arne-W, 2017-07-01:
@AlivaDot is your comment just an alternative solution, or does the solution provided by @AndKe and @RHHoek solve your Problem too?

Arne-W, 2017-07-16:
@AlivaDot Ping :smiley: 
I will close this issue in the next days if nobody stops me....

Arne-W, 2017-07-23:
All right I am closing this now assuming the problem is solved.

closed 2017-07-23 by Arne-W