ArduPilot/apm_planner #992: AP2 on Ubuntu 16.04 (LTS) experience:

AndKe, 2016-11-29:
Some professional users i know, prefers LTS versions.
AFAIK there is only a xenial version, and below is how it behaves on a clean-installed 16.04(.1)
```
$ sudo dpkg -i apm_planner_2.0.24_xenial64.deb 
[sudo] password for andre: 
Selecting previously unselected package apmplanner2.
(Reading database ... 172759 files and directories currently installed.)
Preparing to unpack apm_planner_2.0.24_xenial64.deb ...
Unpacking apmplanner2 (2.0.24) ...
dpkg: dependency problems prevent configuration of apmplanner2:
 apmplanner2 depends on libflite1 (>= 1.4-release-9~); however:
  Package libflite1 is not installed.
 apmplanner2 depends on libqt5script5 (>= 5.0.2); however:
  Package libqt5script5 is not installed.
 apmplanner2 depends on libqt5serialport5 (>= 5.1.0); however:
  Package libqt5serialport5 is not installed.
 apmplanner2 depends on libsdl2-2.0-0 (>= 2.0.4); however:
  Package libsdl2-2.0-0 is not installed.

dpkg: error processing package apmplanner2 (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.13.3-6ubuntu3.1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu5) ...
Processing triggers for bamfdaemon (0.5.3~bzr0+16.04.20160701-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mime-support (3.59ubuntu1) ...
Errors were encountered while processing:
 apmplanner2
```

so, trying to install the missing packets manually
```
$ sudo apt -f install libflite1 libqt5script5 libqt5serialport5 libsdl2-2.0-0 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 libsdl2-2.0-0 : Depends: libsndio6.1 (>= 1.1.0) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
```

drobban, 2017-11-10:
Is this issue still "active". 
Don't know if this is useful info; but I can confirm that install instructions is working for Xubuntu 17.10 "Artful Aardvark"

AndKe, 2017-11-10:
yes, it worked fine for a long time now.

closed 2017-11-10 by AndKe