ArduPilot/apm_planner #997: Tlog Parsing broken

billbonney, 2016-12-07:
There seems to be an issue with the export logs and reading Tlogs as it doesn't account for the timestamp at a guess.

This log is pre changes... missing TIMESTAMP on export...  post changes loading a  blog fails the https://github.com/ArduPilot/apm_planner/pull/995/commits/4036ed036ec2a82dd871efaa05404495045c216a#diff-a25e52e2028f2bd6e3ee989487c963dbR165 check

```
FMT, 253, 76, STATUSTEXT, QBZ, time_boot_ms,severity,text
FMT, 1, 42, SYS_STATUS, QIIIHHhHHHHHHb, time_boot_ms,onboard_control_sensors_present,onboard_control_sensors_enabled,onboard_control_sensors_health,load,voltage_battery,current_battery,drop_rate_comm,errors_comm,errors_count1,errors_count2,errors_count3,errors_count4,battery_remaining
FMT, 136, 33, TERRAIN_REPORT, QiiffHHH, time_boot_ms,lat,lon,terrain_height,current_height,spacing,pending,loaded
FMT, 74, 31, VFR_HUD, QffffhH, time_boot_ms,airspeed,groundspeed,alt,climb,heading,throttle
FMT, 241, 43, VIBRATION, QQfffIII, time_boot_ms,time_usec,vibration_x,vibration_y,vibration_z,clipping_0,clipping_1,clipping_2
RANGEFINDER, 0, 0.56999999300000004, 0.56800001899999997
TERRAIN_REPORT, 0, 381732574, -781266234, 118.435631, -115.64563, 100, 0, 448
EKF_STATUS_REPORT, 0, 0, 4.1096150200000003e-06, 0.0091953035399999997, 0.076508127199999998, 0, 165
VIBRATION, 0, 295750543, 0.0414216928, 0.046523690200000002, 0.070369347900000004, 0, 1, 0
SYS_STATUS, 0, 6421807, 6364463, 6421807, 177, 11956, 19, 0, 0, 0, 0, 0, 0, 94
POWER_STATUS, 0, 5206, 19, 1
MEMINFO, 0, 0, 19376
MISSION_CURRENT, 0, 0
GPS_RAW_INT, 0, 295644000, 381732574, -78
```
**

Arne-W, 2016-12-08:
I can reproduce a similar error with an "AUTOPILOT_VERSION" message: `Number of datafields for type AUTOPILOT_VERSION does not match. Expected:12 got:9`
It is not the timestamp as the handling of them is nearly identical for all log types. I my case the "AUTOPILOT_VERSION" message lacks the last three values. I have to do some further investigations to verify if this is a parsing issue or a log corruption.
[image]
As you can see the last three values are missing - timestamp is ok.

Never the less the LogdateStorage class seems to be too restrictive regarding those corruptions. As the LogdataStorage is able to handle this error (removing the return false at the location mentioned above makes it work) I will change the error handling in the next days. It will be better not to stop parsing but mention this format corruption in the status window presented after parsing.

Arne-W, 2016-12-10:
I did some debugging and I am pretty sure it is a log corruption. So the handling of this should be done by the tlog parser. The only thing which is not clear is how do we want to handle those messages? In my opinion there are 2 possibilities, first we can simply drop them loosing all data of this message. As you can see on the screenshot above some data is missing but the rest seems to be ok. The second way to handle this could be to insert some default data into the missing fields of the corrupt message - "repair" the data.
So what do you think?

billbonney, 2016-12-10:
I'm not sure why they are corrupt but we should do the best we can to extract the data. Maybe we can have a column that marks them with a warning about potential corruption. I was analyzing a log recently and even if the data say 1100 corruption, its the only log I have.

Also I'm trying to plane for MAVLink 2.0 update so fun is coming with newer messages

Arne-W, 2016-12-11:
I think the corruptions are a ardupilot bug. In the case of the "AUTOPILOT_VERSION" parameter mismatch all messages of this type have the same problem. Looks like someone changed the header and forgot to change the rest of the message.
I have a good Idea how to reconstruct a message as good as possible but no idea how to mark those lines - adding a column to those lines breaks the data-model as all rows of one type must have the same signature. Perhaps it is possible to mark those values to deliver additional information in the table view.

billbonney, 2016-12-12:
@Arne-W I think just making it work for now is a priority. But i think it should be noticeable that a data field could be 'corrupt'

Arne-W, 2016-12-12:
done :smile:

billbonney, 2016-12-13:
👍

billbonney, 2016-12-13:
fixed with https://github.com/ArduPilot/apm_planner/pull/998

closed 2016-12-13 by billbonney