ArduPilot/apm_planner #1000: Support new FRAME_CLASS, FRAME_TYPE parameters for Copter-3.5

rmackay9, 2016-12-31:
For AC3.5 we are introducing a FRAME_CLASS parameter and renaming FRAME to FRAME_TYPE so it would be good to update the frame selection screen.

Possible values from FRAME_CLASS are 0:Undefined, 1:Quad, 2:Hexa, 3:Octa, 4:OctaQuad, 5:Y6, 6:Heli, 7:Tri, 8:SingleCopter, 9:CoaxCopter
Possible values for FRAME_TYPE are 0:Plus, 1:X, 2:V, 3:H, 4:V-Tail, 5:A-Tail, 10:Y6B

Here's the equivalent to-do for MP which includes a mock-up of what the screen might look like: https://github.com/ArduPilot/MissionPlanner/issues/1442

Arne-W, 2017-03-22:
@billbonney shouldn't we start implementing this feature? I saw your [comment](https://github.com/ArduPilot/MissionPlanner/issues/1442#issuecomment-269847367) but it seems that nobody tries to find a better solution.
Perhaps we should show only the valid FrameTypes for a selected FrameClass. This would make it impossible to select undefined combinations. The drawback is that we will have to adapt the selections whenever ArduPilot adds or removes some combinations.
I started some implementation tests (very early stage)- see [https://github.com/Arne-W/apm_planner/tree/NewFrameType_WIP](https://github.com/Arne-W/apm_planner/tree/NewFrameType_WIP). At the moment it gives you a blank frame screen when ArduPilot 3.5 is detected. Next steps will be to fill the screen.

Arne-W, 2017-03-24:
@rmackay9 Which FRAME_TYPE has to be selected for Heli and Tri or do they not use the TYPE parameter?

Arne-W, 2017-03-24:
All right - did not take as much time as I thought. If anybody has some time it would be nice if he could test the Branch mentioned above. There is still some clean up to do but the functionality should be complete. The branch should work with all ArduPilot FW versions.

rmackay9, 2017-03-30:
@Arne-W, thanks very much for doing this.

I had a peek at the Copter motors library and Heli and Tri copters both ignore the FRAME_TYPE.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsTri.cpp#L29
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsHeli.cpp#L182
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/system.cpp#L543

I've added some links above but that's not implying we'd expect you to try and figure this all out by looking at the code 'cuz it's not super straightforward.

Thanks again.

Arne-W, 2017-03-30:
@rmackay9  Ah - thanks looks like single and coax are ignoring the FRAME_TYPE too. The documentation [here](http://ardupilot.org/copter/docs/singlecopter-and-coaxcopter.html#singlecopter-and-coaxcopter) only talks about setting the FRAME_CLASS. I will add some buttons for these classes in the next days. @billbonney This should not prevent you from merging.
BTW - Does anyone have some Icons for single / coax / V-tail / A-tail copters we could use for the frame selection? At the moment V- and A-tail buttons have only text - not nice but works. When adding single and coax I can use more text buttons but I think it would be much better to have some good icons.

commit 168751eb (2017-04-02): Frame Config: Added support for ArduCupter 3.5.0 frame configuration. Fixes #1000
closed 2017-04-02 by billbonney