ArduPilot/apm_planner #1001: changes to copter binary names

tridge, 2017-01-12:
Hi Bill,
This is a heads up that binary names for copter builds are changing, which will affect firmware upload
I've created a MP issue describing it here:
https://github.com/ArduPilot/MissionPlanner/issues/1456
Cheers, Tridge
CC: @rmackay9

mboland, 2017-03-27:
Mission Planner can no longer find the binaries for 3.5 beta firmware.
[Forum Issue](http://discuss.ardupilot.org/t/cannot-get-beta-copter-to-appear-in-firmware/16051)

Just wondering when might expect a fix for this?

billbonney, 2017-03-27:
@Arne-W DO you think you can look at this, sorry I don't have time at the moment

Arne-W, 2017-03-27:
@billbonney Yes no problem - I will check that tomorrow. BTW you should take a look at #1000 I made a WIP - if yo like it I will make a PR.

commit a456b593 (2017-04-02): Firmware Upload: Added support for changed file names for Copter firmware. Fixes #1001
closed 2017-04-02 by billbonney