ArduPilot/apm_planner #1002: Move app data directory.

markrjr, 2017-01-15:
It's a relatively minor issue, but could the default directories for app data be moved?

Specifically to `QStandardPaths::AppLocalDataLocation`. 

http://doc.qt.io/qt-5/qstandardpaths.html

This way, APM Planner's app data, logs, and whatnot will be mapped to the platform specific app data directories instead of a user's home folder. Even though [this](https://github.com/ArduPilot/apm_planner/issues/730) issue was closed, it still affects me on macOS Sierra with APM Planner 2.0.24. Moving the directory would probably alleviate the issue altogether and place nicely with each respective platform.

I would be happy to submit a pull request if you'd like, but this seemed like such a small issue that it wasn't worth forking, creating a new branch, etc.

billbonney, 2017-01-16:
You can in the settings change the location of the folder to a preferred location. The default is the users home directory. The proposal you are suggesting would just put the apm_planner directory in 'Documents' subdir on the platform instead.  That is quite a minor change really.  ;)

I don't have anything in my Documents directory, just some fluff before I save most things in a structured way elsewhere.

markrjr, 2017-01-24:
I am proposing that the directory is moved to **`QStandardPaths::AppLocalDataLocation`.**
That is not the platform's documents directory. Specifically, it resolves to: 

macOS - `~/Library/Application Support/<APPNAME>`
Windows - `C:/Users/<USER>/AppData/Local/<APPNAME>`
Linux - `~/.local/share/<APPNAME>`

And I said before, I am still experiencing [this issue](https://github.com/ArduPilot/apm_planner/issues/730) on OS X **which does not allow me to change the folder from settings.**

pepak, 2017-06-24:
It also does not work on Linux.

Arne-W, 2017-07-01:
@markrjr It would be cool if you could verify this bug on current master build. On Linux (Ubuntu 16.04) I can change the folder settings without any issue.

Arne-W, 2018-02-23:
no reaction since last July - closing now.

closed 2018-02-23 by Arne-W