ArduPilot/apm_planner #1005: Graph: Ideas for even better log analysis.

AndKe, 2017-02-06:
AP2 is (by far) the greatest log analysis tool I know. It's speed, smooth navigation is great.

please bear with me for a feature request that would make it even better:
During tuning , writing of POH, gathering performance data or general log analysis, we often need to know an average, or difference during a certain maneuver or time.   

Please see mockup. 
If a checkbutton is selected, two red lines appear, they can be dragged to select area of interest.
the mouseover displays the time between those lines, and we can sea delta (difference) for value during that time frame, and it's average.

Usage examples:
Delta:
- would easily show acceleration (accX) range during normal flight, before some failure that introduced extra vibrations. (and also easily qualtify those , even if the whole graph remain scaled to some other event) 
- find climb rate/descend rate, roll or pitch rate  - when something is happening, let's say roll rate in a turn, or descent rate. 
-find power consumption over xx seconds at xx speed.

Average: 
-easily quantify how much an axis is changing during apparently "flat" logline,  select 10s , see how much variation we have in pitch.
-get proper average throut , used for documenting multirotors,  1kg payload = throut xx  , 2kg=throut xx  - then we know it's maximum capacity vs flight times m vs redundancy reserves. 
-Also, during very smooth hover in still air, properly tuned multi has low variations in throut,  - all rcout.chX should have similar average (and delta)   - otherwise it's a indication for some problem. 
 
The tools, as illustrated, will aid proper log analysis a lot.

[image]

Arne-W, 2017-02-07:
Cool idea - should be done! :+1:

Arne-W, 2017-03-04:
Started to work on this feature, but the classes which are responsible for the graph view need some refactoring cause they get bigger and bigger.
So far I have updated to a new QCustomPlot version and split the the Graph view code in 2 classes - one for online viewing and one for offline log analysis. Next thing will be the cursor...
I thought about a context menu when clicking into the graph to add one or two cursors.
I will tell you when the feature is testable...

AndKe, 2017-03-04:
Great :)

Arne-W, 2017-03-17:
Ok its nearly done :smile: If somebody has some time to waste he could check [THIS](https://github.com/Arne-W/apm_planner/tree/New_LogAnalysisWindow) branch.
New Features:
* Range cursors can be used simultaneously with old cursor.
* cursor enabling / disabling through context menu when clicking in graph.
* cursor moving by select click followed by dragging.
* Y-Axis can be clicked (selected) and zoomed with mouse wheel.
* Splitted life log and offline log windows into 2 separate classes and did some refactoring.
* Migrated to QCustomPlot 2.0.0 beta.

I still have to do some cleanup - the PR will come in the next days.

AndKe, 2017-03-17:
no time to waste, but then, I do not consider this a waste of time :) 

Fantastic - almost there ! 
I failed to find how to move the range cursors (I see vertical red+green line, nothing to grab them by)
-please add value delta.  (difference between min-max you are already displaying)
don't know what "Add simple cursor" is supposed to do.
-finally when displaying data tooltip:  please rearrage so values are in order :  
min , max, delta, avg , cursorposition   
This way, the tooltip will be more pleasant to watch when moving mouse along the graph, as only the last 
The distance between value name and the final numbers on the long tooltip , can be solved by adding a horizontal bar for each two values, or using alternating colors. 

Great work!

Arne-W, 2017-03-17:
Thank you! 
The cursors should be selectable by clicking directly on them. If a cursor is selected you can click on it an move it as long as its clicked. To deselect you have to click somewhere else. Works like the Y-Axis scaling feature.
The simple cursor is the old double click feature. It acts as a link between the Graph and the table view.
Rearranging the values is a good idea - and perhaps I should rename the simple cursor...

AndKe, 2017-03-17:
I see you did not get the edited version, most likely using mail to read it. 
My final addition was also:
"BTW: delta between line numbers in range is displayed with three decimals :)"
Have a nice weekend.

AndKe, 2017-03-17:
On second thought: The doubleclick to log feature could be a green checkbox
below (like in master), or just permanently on, AFAIK, we don't need
doubleclick to be available in graph for any other purpose, so there is
most likely no reason to have an option to toggle it.

AndKe, 2017-03-17:
Final thought: by using a green checkbox to toggle the range selection ,the
UI would have fewer "different places to do things" , and people would
easier discover this great feature.   So I say the green checkboxes are the
way to go.

billbonney, 2017-03-18:
Is there a PR for these changes or is it a WIP?

Arne-W, 2017-03-19:
@billbonney at the moment its a WIP - we are still discussing some behaviour. I think it will become a PR in the next days. If you want to test it [THIS](https://github.com/Arne-W/apm_planner/tree/New_LogAnalysisWindow) is my WIP branch.
@AndKe I made some changes would be nice if you could have a look.

AndKe, 2017-03-19:
@Arne-W 
Very nice !  works great.   
One more thing while you are at those files, the default loaded log does not display the grid with data, a new user would not be aware of it's existence.   Please change the GUI so it reveals at least 3-5 lines if data.

Arne-W, 2017-03-19:
Done :smile: - I will create a PR

AndKe, 2017-03-19:
OOPS ! : please see below: 

Altitude minimum 174.11 ,max 174.75, and the arerage is 261 ?  :)

[image]

Arne-W, 2017-03-19:
Found and fixed it - thanks :+1: - a typical off by one! Anything else :smile:

AndKe, 2017-03-19:
unrelated; are you able to reproduce this https://github.com/ArduPilot/apm_planner/issues/646#issuecomment-221203527 ?
(just getting a bit desperate on that one)

Arne-W, 2017-03-29:
As feature is completed and merged I will close this.

closed 2017-03-29 by Arne-W