ArduPilot/apm_planner #1006: Graph display incorrect Mode from Radio Button under graph window

mboland, 2017-02-13:
With the Radio button under the graph window all modes are displayed as "Stabilise by Radio" regardless of the actual mode.

[image]

When the Mode/Mode log entry is selected the Mode is displayed correctly.

[image]

[image]

Arne-W, 2017-02-13:
I will check that...

Arne-W, 2017-02-13:
All right - I can not reproduce this issue. @mboland could you please upload the logfile which causes the issue?
Did I understand you right - you opened a .bin file with the Mission Planner, exported the log with Mission Planner to a .tlog file and than opened this exported tlog with the APM Planner which leads to this misbehaviour of the Mode arrows? Opening the bin file directly with the APM Planner does not show this issue?
I have the suspicion that there is something fishy with the Mission Planner export.

mboland, 2017-02-13:
I have mission Planner on a WinLapTop in the workshop which I use to download the logs.
Running latest beta version and it converts the .bin to .log on download.
I use MacOS in the office for analysis and initial setups.
APM Planner has the best graphing facilities.

So, I will usually just select the .log file to open in APMPlanner and I get what you see.
Using the .bin file APMPlanner opens  much quicker but if Mode/Mode is selected a numeric graph is produced rather than labels.
[image]
Guess I will just use the .bin files from now on.

Arne-W, 2017-02-14:
Ah okay - looks like a Mission Planner export issue.
As we have reimplemented the the log parsing and the internal data storage we changed the behaviour of the labeling. The labels are only supported for the 4 message types which have a checkbox below the graph. At the moment we support MODE, ERROR, EVENT and MESSAGE events to have labels. All data that can be selected through the tree view on the right will produce a graph but no labels anymore.
I thought it could be sometimes useful to have both representations. Only measurements which contain a string as value can not be plotted as a graph - they only have the label-arrow representation.

closed 2017-02-14 by mboland