ArduPilot/apm_planner #1007: Info view: missing data

AndKe, 2017-02-16:
GS, AS, GPS sats and GPS fix is not correctly updated in info view:
Screenshot says it all:

[image]

Arne-W, 2017-02-18:
Not really a bug - the name of the mavlink data changed and your Info View has stored its configuration in the ini file. So it loads the wrong configuration on every startup. You could easily remove the "GCS status gps" values and replace them with the "GCS GPS gps" vaues which can be found there as soon as ardupilot hat transmitted them.
I will make a PR which adds a migration method to change these names automagically.

billbonney, 2017-02-18:
On a new version update it wipes these files 'fixing' the issue. That said this problem is from the feed names on that reset see 
see UASQuickView
```
//If we don't have any predefined settings, set some defaults.
    if (uasPropertyValueMap.size() == 0)
    {
        m_columnCount = 2;
        valueEnabled("GCS Metric.Alt MSL (m)");
        valueEnabled("GCS Metric.Alt REL (m)");
        valueEnabled("GCS Status.Roll (deg)");
        valueEnabled("GCS Status.Pitch (deg)");
        valueEnabled("GCS Status.Yaw (deg)");
        valueEnabled("GCS Status.Climb (m/s)");
        valueEnabled("GCS Status.Voltage (V)");
        valueEnabled("GCS Status.Current (A)");
        valueEnabled("GCS Status.Battery (%)");
        valueEnabled("GCS GPS.GPS Fix ()");
        valueEnabled("GCS GPS.GPS Sats (n)");
        valueEnabled("GCS GPS.GPS HDOP ()");
        saveSettings();
    }

```

Arne-W, 2017-02-18:
Ahh - you mean it is the missing version change that leads to this issue?! So I can drop my branch with the migration method?!

billbonney, 2017-02-18:
@Arne-W I think you can ignore migration, unless you want to do it. It's quite complex to maintain and 'fragile'. We adopted the 'reset-all' approach on version update 

That said the code above should list valid values for the latest version ( I changed some I think, hence why its a problem)

Arne-W, 2017-02-18:
@billbonney The reset all approach is the easiest way to handle these issues - I just forgot the reset on version change. I will drop the PR.
@AndKe So you read it all - just delete you ini file and everything will be perfect :smile:

AndKe, 2017-02-18:
Yes, I see, - (and it works fine)  :)

Arne-W, 2017-03-21:
As this was not really a bug and was solved - could be closed

closed 2017-03-21 by AndKe