ArduPilot/apm_planner #1010: Blank EKF and VIB monitor screens

RHHoek, 2017-03-04:
on request of @dcarpy I open this issue.
...For me this problem still isn't solved ( on Ubuntu 16.04/16.10). Lately I built AP2 from the latest source.
Is this problem solvable?
@carpy offered a solution more than a year ago, but I don't understand how to implement this solution.
Is there something I can do? Is this perhaps a video driver issue? Is there a simple Qt-test I can run?....

see discussion: http://discuss.ardupilot.org/t/blank-ekf-and-vib-monitor-screens/7068/7

Arne-W, 2017-03-21:
Wasn't this issue fixed with #1011 ?

dcarpy, 2017-03-21:
Not really.  I had them install qml-module-qtquick-controls and the problem (blank graphs) went away.
This could be closed as "Not a bug" or make a comment, "Install package qml-module-qtquick-controls."

RHHoek, 2017-03-22:
This issue, I think, can be closed by updating the README.md (required packages for linux (and OSX?)) and perhaps .travis.yml and debian/control

RHHoek, 2017-07-11:
The README.md still needs an update.
The line:
sudo apt-get install qml-module-qtquick-controls
needs to be added to the section "Linux, Install the required packages:"

Arne-W, 2017-07-12:
@RHHoek - you are right - I will update it in the next days....

commit 6bca0edd (2017-09-07): Updated Readme.md Fixes #1010

* Added qml-module-qtquick-controls to the Ubuntu package requirements.
closed 2017-09-07 by billbonney