ArduPilot/apm_planner #1012: APMPlanner2 did not start.

mrprole, 2017-03-12:
Windows 7 Pro, 64
Here is what i got with Dr. Mingw:
apmplanner2.exe caused an Access Violation at location 670C7D9D in module ig7icd32.dll Reading from location 00000018.

Registers:
eax=07cd5f18 ebx=00000000 ecx=00000000 edx=07cd5f18 esi=07cd5ec0 edi=05243460
eip=670c7d9d esp=0039f44c ebp=0039f454 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00210206

AddrPC   Params
670C7D9D 07D799C8 07C8F6E8 00000000 ig7icd32.dll!RegisterProcTableCallback
670C7EE2 10000001 0000027A 000001C5 ig7icd32.dll!RegisterProcTableCallback
672BE78D 07C7FE64 0039F4FC 670C75D2 ig7icd32.dll!RegisterProcTableCallback
670C50A3 07BB8C90 07C7FE64 07D91C40 ig7icd32.dll!RegisterProcTableCallback
670C75D2 07BB8C90 07D91C40 07D91C40 ig7icd32.dll!RegisterProcTableCallback
670C7AF8 774A22C0 00000000 66F9933C ig7icd32.dll!RegisterProcTableCallback
67034BD0 07BB8C90 00000001 00000000 ig7icd32.dll!RegisterProcTableCallback
6736DEDD 5F011042 07D91C40 67749028 ig7icd32.dll!DrvSetContext
67749422 00010000 00000003 00000000 opengl32.dll!wglSwapLayerBuffers
677495E9 5F011042 00010000 0039F78C opengl32.dll!wglMakeCurrent
67CE09B4 0039F78C 07CC67A8 07C3FF70 qwindows.dll
67CDF8A0 0039F78C 0039F768 00000000 qwindows.dll
67CE0700 0039F778 67CE048B 774AE013 qwindows.dll

mrprole, 2017-03-19:
OK. I solved it. I am using Thinkpad X230T with docking station. Some time ago i installed **[Lenovo USB Display Driver](http://pcsupport.lenovo.com/ua/en/products/LAPTOPS-AND-NETBOOKS/THINKPAD-X-SERIES-TABLET-LAPTOPS/THINKPAD-X230-TABLET/downloads/DS102800)**. Just uninstalled it and APMPlanner2 started without any problem.

closed 2017-03-19 by mrprole

Arne-W, 2017-03-19:
Nice to hear you solved the problem, and thank you for informing us!