ArduPilot/apm_planner #1013: Please fix minor planning issues:

AndKe, 2017-03-17:
When dragging a WP, we can see coordinates.
Please add distance from previous WP, and gradient.

[image]

Tooltip over an WP does not calculate heading, but it would be even more useful to see:
-distance from previous WP to current WP
-distance from first to current WP
-gradient from previous WP

[image]

hdino, 2018-07-15:
I agree that these tooltips are the best places to provide some additional information. Before I try to implement this, we should clarify some questions:
1) How shall we define distance? 2D or 3D? If a gradient is given, I think 2D should be our choice.
2) How to express the gradient (or is slope or grade the better word for it?)? Altitude difference per 2D distance, so [m/m]? Or expressed in percent? Or should we just show the altitude difference in meters?
3) If we add both the distance from the previous WP and the gradient to the drag tooltip, the text box would become quite big. (We could add a second line though.) Therefore, I thought about adding something to the general config screen to make this configurable. Idea 1: Add a checkbox to enable/disable this additional information. Idea 2: Add a combobox to choose from coordinates/distance+gradient/coordinates+distance+gradient. Idea 3: Make it completely customizable, i.e. add a textbox where you can type something like "%lat %lon\n%dist2d %gradient". This would be great especially for the non-drag tooltip.
4) The distance from the first to the current WP should be the 3D distance, in my opinion. Alternatively, it could be the 2D distance with the total positive change and the total negative change in altitude (e.g. Distance from 1st WP: 180 m. Total ascent: 80 m. Total descent: 60 m).

AndKe, 2018-07-16:
2D is enough for most planning , for most missions, the actual ground distance is simply not that critical, when we have one huge uncertain factor as head/tailwind differences.
But 3D calculation of the whole route may be nice, and does not require much more.
The more the better, when it comes to information, for me, multiple lines sounds good, or . a dedicated toolbar/statusbar area - or, any of your suggestions :)
The gradient is best expressed in % , together with ground distance, it's foolproof information.