ArduPilot/apm_planner #1014: Futaba T8FG + R6208SB + Pixhawk

razvanpaul, 2017-03-18:
Hello folks!

Firstly, thank you very much for this open source software. It's really amazing.

Unfortunately, I have an issue. I've used APM Planner 2.0 on a Mac successfully until the moment, but I got stuck into one issue. The radio signals aren't received at all by the software, I get nothing:

[image]

The usb connection does work (I have calibrated the accelerometer and the compass and I receive GPS  data), the LED on the R6208SB receiver is green and I have the "MULT" channels selected on the Futaba Radio. Do you know what could cause the bug on APM Planner?

Arne-W, 2017-03-18:
For me it looks like there is an connection problem between the receiver and the flight controller. Are you sure your SBUS connection works?

mboland, 2017-03-18:
I have experienced this in the past with the Futaba SB connections, although I am not sure I can blame the Futaba, it's just what I was using. 
It is very random in nature and I could never pin it down to an exact cause.
Try:
1. Restart everything, PC, Mission Planner, radio, APM and try again.
2. Install latest beta version of Mission Planner, there are always changes going in.
3. Reflash the firmware on the APM. I never really thought this was a fix but sometimes it would just work afterwards.

My feeling was it was always an issue with Mission Planner that it just seems to gag on something.
At present it seems to do it when you have multiple compasses to calibrate.
Have you tried other GCS's?
QGC or APM Planner?

billbonney, 2017-03-18:
please use http://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0 for support. if it's a bug, we will reopen this again. Thx 👍 

PS: don't forget to mention autopilot HW you are using as well

closed 2017-03-18 by billbonney