ArduPilot/apm_planner #1015: apm_planner.pro won't compile. Errors in Mavlink type names

badaniels, 2017-03-18:
I just tried to compile the APM Planner app for mac and I get a number of errors when trying to use the `make -j8` 
instruction. however when I try to compile with just the `make` command I only get 3 errors. all of these errors are with mavlink message types. 

I am trying to compile a version of APM Planner with my own Mavlink messages to show a telemetry reading of some additional sensors that I have added, however when I try to compile the Master version I get the same 3 errors. 

Raw error readings from the terminal are 

`In file included from src/main.cc:31:
In file included from src/QGCCore.h:38:
In file included from src/ui/MainWindow.h:46:
src/uas/UASInterface.h:691:32: error: unknown type name 'mavlink_compassmot_status_t'
    void compassMotCalibration(mavlink_compassmot_status_t* compassmot_status);
                               ^
In file included from src/main.cc:31:
In file included from src/QGCCore.h:38:
In file included from src/ui/MainWindow.h:77:
In file included from src/ui/SlugsHilSim.h:44:
In file included from src/comm/LinkManager.h:52:
In file included from src/comm/UASObject.h:7:
src/comm/VehicleOverview.h:401:52: error: unknown type name 'mavlink_vibration_t'; did you mean
      'mavlink_radio_t'?
    void parseVibration(LinkInterface *link, const mavlink_vibration_t &vibration);
                                                   ^~~~~~~~~~~~~~~~~~~
                                                   mavlink_radio_t
libs/mavlink/include/mavlink/v1.0/ardupilotmega/./mavlink_msg_radio.h:15:4: note: 
      'mavlink_radio_t' declared here
}) mavlink_radio_t;
   ^
In file included from src/main.cc:31:
In file included from src/QGCCore.h:38:
In file included from src/ui/MainWindow.h:77:
In file included from src/ui/SlugsHilSim.h:44:
In file included from src/comm/LinkManager.h:52:
In file included from src/comm/UASObject.h:7:
src/comm/VehicleOverview.h:402:58: error: unknown type name 'mavlink_ekf_status_report_t'; did you
      mean 'mavlink_statustext_t'?
    void parseEkfStatusReport(LinkInterface *link, const mavlink_ekf_status_report_t &report);
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                         mavlink_statustext_t
libs/mavlink/include/mavlink/v1.0/ardupilotmega/../common/./mavlink_msg_statustext.h:10:4: note: 
      'mavlink_statustext_t' declared here
}) mavlink_statustext_t;
   ^
3 errors generated.
make: *** [build-release/obj/main.o] Error 1`

Arne-W, 2017-03-18:
On linux there is no problem with building the current master even with make -j8. There must be something with your configuration.

badaniels, 2017-03-19:
the errors I get with make -j8 are:-
`error: unknown type name 'mavlink_vibration_t'
error: unknown type name 'mavlink_ekf_status_report_t'
fatal error: '../../src/multimedia/qtmultimediadefs.h'`

which appear to be the same ones as with just make, the errors are just repeated. been trying to find out what they should be declared as, but cant find them anywhere. If some one know what they're supposed to be and where I can get the `multimediadefs.h` file that would be a massive help. 

I've looked thought the files on here and cant find it anywhere. also wondering if that missing file is the root of the issue.

billbonney, 2017-03-19:
this is not an issue with APM Planner 2.0. post here for support http://discuss.ardupilot.org/

make sure you build the mavlink libraries correctly from the mavlink github ie. 
`python -m pymavlink.tools.mavgen --lang C --wire-protocol 1.0 message_definitions/v1.0/ardupilotmega.xml`

closed 2017-03-19 by billbonney