ArduPilot/apm_planner #1016: cannot view logs

govsux, 2017-03-20:
Am getting this a lot with 2.0.20
https://drive.google.com/open?id=0B3MgBBZGOf88N21INUdGSEROT0U

Arne-W, 2017-03-20:
Did you find the reason for the corrupted data? Is it really a problem with the Log, or a problem while parsing the data?
If you are sure it is a problem in the logfile - everything is OK, if not it would be great if you could upload the log that lead to the result above, I will check then.

govsux, 2017-03-20:
I'm not really sure I just noticed that I could in fact get the bin file to graph okay although I did notice it would show when a mode was changed but not indicate what modes they were so something is going on?
https://drive.google.com/open?id=0B3MgBBZGOf88d1Q4akh0a3lsYTg

Arne-W, 2017-03-20:
Ahh thank you.
I did some testing, and its definitely a log problem. The EKF1 message contains only a valid Timestamp field, all other values are set to 0.0 except EKF1.PD which jumps sometimes to incredible high values. Some of them are filtered out by an "is Not a Number" "is not infinity" check. I seems like ArduPilot does not fill this values any more - perhaps due to migration to the NKF which has proper values and nor errors.
To summarize - everything is OK. You can ignore those EKF1 problems as the messages do not contain any useful data. It is definitely not caused by APMPlanner, its an ArduPilot FW problem.

govsux, 2017-03-20:
I see okay. Where you able to see when modes were changes and what they were?? I wasn't.

Arne-W, 2017-03-20:
You must enable the "Show MODE" checkbox right below the graphing window. All other features of the graphing worked as usual even the plotting of the EKF1 data works - slightly useless cause all values are 0 - but works.

govsux, 2017-03-20:
ah yes... thx

govsux, 2017-03-20:
I see mode 6 mode 2 but not the actual mode aka alt hold etc. Anyway... I guess this should be closed as the 'cannot view logs' isn't a real issue at least not with apm p?

Arne-W, 2017-03-21:
Looks like you have an old version of APM-Planner. Those IDs are shown if APM cannot detect the MAV Type (aka Quadrotor/fixedwing/rover....). Since Ardupilot 3.4 the old detection algorithm did not work anymore due to parameter renaming.
This was fixed with 82ee43d09f3c76c315e4045760cc1aa41cf69a52 on 04.06.2016.
So you should update.

Arne-W, 2017-03-29:
No response - seems to be solved.

closed 2017-03-29 by Arne-W

govsux, 2017-03-30:
Yes you are correct an update did indeed solve that 'no mode' issue

Arne-W, 2017-03-30:
You are always welcome - thanks for the feedback :+1: