ArduPilot/apm_planner #1023: Graph: where did "filter" go ?

AndKe, 2017-04-01:
I miss the "Filter" feature that filtered the datagrid lines by MSG type.
Was it removed intentionally, or by accident when adding new graphing features  ?

Arne-W, 2017-04-01:
Which "Filter" do you mean?
[image]
This one?

AndKe, 2017-04-02:
Yes , do you see it in current master ? (I don't)

Arne-W, 2017-04-02:
As the current master has currently changed you should try again - and yes - I can see it.

AndKe, 2017-04-02:
Very strange, again, I just built clean from master, (2.0.20-149-g5334dbe)

here's how it looks - and no, increasing width does not reveal the button I miss. 
[image]

what I do is open apmplanner2 ,click "graphs", "open log", and load a  .BIN
I've also deleted the ini file.

Arne-W, 2017-04-02:
Hmm strange - when I resize the graph window to the smallest possible size it looks like this:

[image]

I can not make it smaller.
I think I can make a test version with resized buttons - I will inform you!

AndKe, 2017-04-02:
Please note that it's not about "small window",   I have normal size window
just the same, no amount of resizing (or fullscreen window) reveal that
button.

mboland, 2017-04-02:
Is this in Daily build yet?
Or do I need to compile Master?
I have 3 different screen sizes from largest to small I can test on.

Arne-W, 2017-04-02:
Yeah - it should be. Would be nice if you could have a look.

mboland, 2017-04-02:
No Filter button visible.
Details:
AMD FirePro D700:
  Chipset Model:	AMD FirePro D700
  Type:	GPU
  Bus:	PCIe
  Slot:	Slot-2
  PCIe Lane Width:	x16
  VRAM (Total):	6144 MB
  Vendor:	ATI (0x1002)
  Device ID:	0x6798
  Revision ID:	0x0000
  ROM Revision:	113-C3861J-687
  Automatic Graphics Switching:	Supported
  gMux Version:	4.0.11 [3.2.8]
  EFI Driver Version:	01.0D.687
  Metal:	Supported
  Displays:
Thunderbolt Display:
  Display Type:	LCD
  Resolution:	2560 x 1440
  Pixel Depth:	30-Bit Color (ARGB2101010)
  Display Serial Number:	C02Q10LVF2GC
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Automatically Adjust Brightness:	No
  Connection Type:	DisplayPort
Thunderbolt Display:
  Display Type:	LCD
  Resolution:	2560 x 1440
  Pixel Depth:	30-Bit Color (ARGB2101010)
  Display Serial Number:	C02Q10NJF2GC
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Automatically Adjust Brightness:	No
  Connection Type:	DisplayPort
[image]
[image]

Arne-W, 2017-04-03:
@AndKe you must enable the table view in order to see the filtering options. No table no filter. Sorry I forgot this.
@mboland thanks for the support - but you have to load a log and enable the table view in order to see the filter button.

AndKe, 2017-04-03:
@Arne-W  oops ! - I just dragged the horizontal bar to show the table/grid, which does not spawn the filter button.

All green checkboxes except "hide table view" are "check to enable a feature"  ,  the "hide" implies reversed logic : uncheck to enable a feature.

It would be nicer to invert it, having label "Show Table View"   - also, whenever the horizontal split line between graph+checkbozes and table view is moved by mouse, "Show Table View" should be ticked.

mboland, 2017-04-03:
Found it, and I didn't even know it existed.
I usually have the table view hidden but with a log loaded and the table in view, there it was.
[image]
[image]
[image]

Arne-W, 2017-04-04:
@mboland Thanks for testing, as I do not have an OSX to test it is always nice to see it works on OSX as expected.
@AndKe Regarding the checkboxes - You are right - all boxes are "show if checked" except the table view one. I will change the logic.
I didn't even know that it is possible to make the table visible by dragging the horizontal split line. I will have a look.

AndKe, 2017-04-04:
@Arne-W you said: "I didn't even know that it is possible to make the table visible"

I have suggested earlier to reveal a few lines at all time, just so people could discover the grid view, as I were unaware that it had a button. - To you that must have looked like gibberish :)

Looking to show grid, my brain completely ignored a checkbox starting with "hide..."  and so I never used that button.   Funny how differently different users may use/experience the same software.

AndKe, 2017-04-04:
Another solution, may be to show a few lines of the grid by default (so user may discover it, but hide it), and not to have a checkbox for it at all.

Arne-W, 2017-04-04:
Hmmm - the drawback is that if the user hides the table view completely - the filter button is still visible and clicking on it has no effect (as table view is hidded) except that it changes its text from show to hide.
At the moment I have not Idea how to detect that the user has hidden the complete view - in oder to show or hide the filter button. I can detect the user is moving the horizontal "splitter" but I can't detect if its completely hidden.

AndKe, 2017-04-04:
If every graph window started with let's say 10% are a for the table , user would know about it's existence. Anyway that was just a suggestion, not necessarily a good one :)

Arne-W, 2017-04-04:
What Do you think about this:

[image]
This is the window directly after loading. The "Show Table View" checkbox is removed and the "Show Filter" button is renamed to "Table Filter" and moved next to the "Reset Scaling" button.

[image]
This is the window if the user resizes the table until it is invisible. The "Table FIlter" button gets disabled as soon as the table view gets completely hidden.

If you want to test - here is my branch: [https://github.com/Arne-W/apm_planner/tree/LogAnalysis_filter_WIP](https://github.com/Arne-W/apm_planner/tree/LogAnalysis_filter_WIP)

AndKe, 2017-04-04:
Looks very good. I like it.

commit e43ab2f2 (2017-04-13): GraphView: Fixed UI inconsistency regarding table filter and reorganized some UI parts. Fixes #1023

 * Moved filter button to the middle
 * removed "Show Table View" checkbox as the view can be removed using the splitter
closed 2017-04-13 by billbonney