ArduPilot/apm_planner #1026: Wish list for graph module :)

lvale, 2017-04-05:
Hi.
I was talking with @billbonney about the great work that has been done on the graphing and my wish list is:

Independent automatic scaling of axes per measure. i.e. measuring Amps and Altitude on the same graph would have the scale from say 0 to 50A  to Amps and Altitude from 0m to 120m 

Possibility of saving graphing settings: I usually do a common graph with a series of values and I repeat those same settings for every log I analyse.

Possibility of independent adjusting scale/offset/LPF for a graphed property.

Possibility of comparing and graphing values from 2+ different logs. ie, compare flight1 altitude with flight2 altitude, etc

FFT :)

Arne-W, 2017-04-05:
@lvale nice to hear you like the changes :smiley: 
Cool ideas for improving the log analyzing - perhaps I can realize some of them. But first some questions:
 * Independent automatic scaling: At the moment the graph scales to min / max of the selected measurement to make sure everything is visible. So all Graphs are scaled independent. If you want them to have the same scaling you can group them in the "advanced scaling" window. So from my point of view it is already there.
 * Saving settings: Do you mean one setting that is restored if you open the window again or several selectable settings?
* Possibility of independent adjusting: You can use the "advanced scaling" button and / or you can click on the Y-Axis on the left side. I should be selected then. If you use you mouse wheel when the mouse is directly over the selected axis it will change its scaling. This can be done for each axis independently. LPF Low Pass Filter? Selectable cut-off?
 * FFT: Select one measurement and add FFT (if enabled) to same graph as long as only one measurement is selected? Would be like the horizontal grid lines.

lvale, 2017-04-05:
@Arne-W 
 about the saving of settings. Imagine that one could save a graph setting that would combine RCOUT1, RCOUT2,RCOUT3,RCOUT1 and CURR with some specific settings applied to each scale. So one would open a log and then apply a "saved setting" of the graph. A bit what MP does with that list of graphs, although here we would also have the transformations of scales, LPF's, offsets also applied to each graphed value.

Arne-W, 2017-04-07:
@lvale I made a proof of concept for the settings storage [https://github.com/Arne-W/apm_planner/tree/ReapplySettings_Proof](https://github.com/Arne-W/apm_planner/tree/ReapplySettings_Proof).
In the middle of the LogAnalysis window you will find 2 new buttons "Store Settings" and "Apply Settings". You can set up a perfect view onto your data and press store, after that you can apply those settings to every other LogAnalysis window by pressing apply. At the moment only one setting can be stored, if you want more than one - let me know.

It would be great if you could check it and give feedback whether this is the feature you were looking for!
Furthermore it would be great if you have an Idea for the UI, as the 2 buttons in the middle are a mess but for a proof of concept it was enough :smiley:

AndKe, 2017-04-07:
Oh yeah ! - this is great  !
I find myself often browsing thru, and selecting different sets of data to analyze something, like current, voltage or thrust level vs vibrations.
-selecting all motors to compare for similar/equal thust at hover (indication of a bad motor/prop)
and many more typical diagnostic configurations. 
" if you want more than one - let me know."
now you know  :) if you add a droplist with selections to choose from,  that would be great !

I would contribute with a bunch configs myself. 

I see you store colors, that's great ! , as sometime colors may be too similar. 
Also, unless too much work, consider storing "advanced scaling" settings  - advanced scaling groups - it is useful for , for example motors: one would scale all as a group, then it's easier to spot one that is under-performing.    

Finally, there's an old bug in "advanced scaling , groups"  - to reproduce:  plot voltage ("auto axis" checked), your graph will be between 24 and 26volts. now add voltage to a group(only item in group), refresh .. suddenly the voltage is plotted in range 0...26  

BTW: while you are editing those files: "auto scale" is a better, more describing name for the function than  "auto axis"

Arne-W, 2017-04-14:
As there was a lot of progress the branch above should not be used anymore.
Here is a new one: [https://github.com/Arne-W/apm_planner/tree/Analysis_Presets_WIP](https://github.com/Arne-W/apm_planner/tree/Analysis_Presets_WIP)

There are some new features:
 * UI reorganisation of Log Analysis window. We have a menu bar now.
 * Export buttons moved to the menu bar.
 * New "Analyzing Preset" menu which allows to store the current View and its scaling.
 * Support of several different presets which can be loaded and stored to different Files.
 * File format is "ini" format which makes it editing easy.
 * New "Save View" and "Apply View" buttons making it possible to give one special setting to all othe Log Analysis windows regardless whether they are already open or not.

AndKe, 2017-04-16:
looks good, it seems like you plan to use multiple preset files (.ini)  which will be loaded by default ? - is the plan to have one with the apmplanner, and user should make a custom file ?
Maybe it's better to always load a standard file and the user file, and just save any custom user views to the user file. So that the user don't need to thing about filename - and his presets are auto-saved when exiting.

Arne-W, 2017-04-16:
I thought the best solution to deal with different parameter names in different ArduCopter / ArduPlane... releases is to have several preset files which can be selected by the user. So my idea was to have some default preset files for different ArduXXX releases. Moreover the user should have the possibility to enhance the default ones, to create new sets and perhaps share their them with others. This is important as the user can adopt the settings files if no one has the time to adopt the presets in the repository.
At the moment the last used preset file is loaded on startup - I think loading always an default preset leads the problem that you have to handle many different FW types (plane/copter/rover) and versions in one preset file - nearly impossible without grouping.

AndKe, 2017-04-16:
i see :)

Arne-W, 2017-04-18:
Again an update on my WIP branch.
The manual / auto scaling is now stored in the presets. If no "Y_AXIS_MAX" / "Y_AXIS_MIN" is found in the ini file the graph will be set to autoscale. Grouping is not supported by the presets at the moment.

Arne-W, 2017-04-19:
All right - I think it is done now. I added the graph grouping feature to the presets. The feature should be complete now. Anyone who wants to test - its still the same WIP branch mentioned [here](https://github.com/ArduPilot/apm_planner/issues/1026#issuecomment-294186401).
@lvale As you have "suggested" this feature it would be cool if you could have a look.

AndKe, 2017-04-19:
a few things I've noticed:
1:selected items, "add view to preset" , gave it name:  close graph = preset is lost (no warning)
2:selected items, "add view to preset" , gave it name, "save presets" , close graph ; preset is lost (no warning, no question for filename)   
3:  - saved next preset "as" file, "test"   - but the suffix ".ini" is not added automatically, so it wont be visible to user.  
4: auto axis/axis range is restored just fine :) 
5: please load last preset.  - and somewhere show which preset file is selected.  (as is, if I load AC3.2presets.ini  - I can't be sure later if it's that, or another version. )

great work, BTW: what are the  "Save view" and "apply view" supposed to do ?

lvale, 2017-04-19:
@Arne-W - added to my to-do list. Will compile later tonight

Arne-W, 2017-04-19:
Ahh thank you @AndKe 
Your first points are already solved :smiley:
Third and fifth will be solved - this evening or tomorrow

The "Save View" and "Apply View" button are for convenience. You don't have to give a preset name, you don't need a filename and if you have opened several analysis windows you can press save in window A and apply in window B without selecting a preset file or a preset name. All loganalysis windows automatically share the "Save View".

Arne-W, 2017-04-20:
Got it - Third and fifth are solved too. Name of the loaded preset will be shown in the window title.
Happy testing :smile:

AndKe, 2017-04-21:
All I would like to know if you think the preset ini format is frozen now,
or if there will be changes that make my mess up presets I make.
(to me, it seems so good, I don't really want to "just test" it more, just
do useful things as test)

lvale, 2017-04-21:
Had some time to explore last night's build.

Some comments:

Changing scaling from auto to manual isn't "respected", ie changing the scale of the RCOUT's then the check box gets checked again.
Assigning colors to values isn't working:

[image]
[image]
[image]

The filtering of data would be a nice to have to "smooth" the data, if we want to check the general tendency of the analysed value and the "spiking" makes it harder..

[image]
[image]

The loading of the parameters is a very nice feature

AndKe, 2017-04-21:
@lvale as for "smoothing"   - you have better than that already(Thanks to Arne-W):
Use the range cursors, and enable values under mouse - you'll see average for selected time period.  
I don't know even one occasion where I would like so hide "spikes" in a log  - quite opposite, I would consider such feature "dangerous" to my understanding of the data.    Just a few days ago, I helped somebody and accidentally discovered bad airspeed sensor that peaked to ~50m/s at random intervals.

lvale, 2017-04-21:
@AndKe it depends on how "aggressive" you go. The range values are great for single log, but it doesn't make sense to "save" it on the settings but imho a LPF filter can be used  as a setting, so that you can quickly get differences between different logs, until we can get more than one log at the same time :)
Same log file with different LPF settings

[image]
[image]
[image]

AndKe, 2017-04-21:
Comparing logs I get, but please give me one scenario where you actually are using LPF, because you need it - I get the picture above, it is kind of easier to the eye, but then hides some sudden output. 
I mean as a workflow, how/when exactly is that nice to get away with it - maybe it's something I will learn to appreciate :)

kd0aij, 2017-04-21:
I use the LPF in FlightPlot to look at raw accelerometer data, since a complementary filter (like DCM) responds only to relatively long-term trends.

lvale, 2017-04-21:
The RC_OUT+CURR+ALT is a usual setting, because I have to evaluate the "general" output of the motors given some other variables. Example: the evolution of each motor as the drone "climbs" vertically, and is affected by turbulence.

Arne-W, 2017-04-21:
All right - thanks for the feedback :+1: .
* @AndKe format of actual branch is frozen now.
* Colors - don't work - I will fix that
* Auto / Manual Scaling - As long as a plotted value is in a group you can not switch it to manual. It is a bit tricky to select the min / max value of 5 graphs in a group - which of the group members determines the values?
* You can open as much logs at the same time as you want, if you want to compare them. That is was the "save view" and "apply view" buttons are for. Open first log, find good scaling, press save, open second log, press apply - tadaaa 
* Filtering seems to be nice feature but is one of the next steps :smiley:

AndKe, 2017-04-21:
Ta-Daa !!  indeed, I would suggest some tooltip description of the buttons,
otherwise only a few hardcore users will ever know what they are for, and
the average user will never think of them as a comparison tool.

Arne-W, 2017-04-23:
Again an update on the WIP branch.
* fixed the colour bug
* fixed a bug where saved groups defaults to group A
* added some tool tips - would be great if someone could review them.

kd0aij, 2017-04-23:
The tool tips made sense to me, and I found it useful to be able to apply the same view to a second analysis window.

Saving/loading views to a presets file seems to work OK, except I couldn't find a way to save an edited view without first deleting it from the presets.

Also, when you put two graphs in the same group, both show as "Auto Axis", but it seems that the Min/Max range for the first graph actually sets the scaling for both. 

Question: Is there a way to "pan" the Y-axis of a graph? (perhaps by dragging the Y axis) That would be much easier than opening the scaling menu and typing min/max values.

Arne-W, 2017-04-23:
Thanks for the feedback - it should be possible to click the y-axis and scale it by mouse wheel.

kd0aij, 2017-04-23:
I just noticed that the mouse wheel Y-axis scaling "zooms" about the Y-axis value under the mouse cursor. So it's possible to achieve a translation with two successive scale operations.

lvale, 2017-04-29:
Graph colours change when you choose the "use time on x axis" option
[image]
[image]

The y axis scale take different space (on the x) according to the values of the axis.
Also would be better that each y axis could take less space horizontally because at times the room the axis scale take is more than the graphs :)
[image]

lvale, 2017-04-29:
updated screenshots from test build
[image]
[image]

if a user creates a group to graph (imagine RC_OUTS) it should only display one y axis for that group

[image]

kd0aij, 2017-04-30:
To correlate the graph with the KML display in Google Earth, we need the option to display GPS time on the x axis.

AndKe, 2017-05-01:
yes, and: the absolutely nicest thing for log analysis, would be to be able to see the data in AHI & map right in AP2...  maybe click a spot in log could update a map & AHI with closest/next messages that contain data needed.

Arne-W, 2017-05-01:
Nice to hear about cool an good ideas - but the last feature has still to be pulled in and some improvements are still on the list. Moreover I will start a new job tomorrow and I will not have as much time as I had in the last 8 weeks.
All I wanted so say is that I have to reduce my effort and so progress from my side will slow down - sorry. Nevertheless I will try to realize your ideas and make APM-Planner a better tool.

AndKe, 2017-05-01:
@Arne-W wish you all the best @ new job, sorry to see you getting less time to spare :)

Related to the current preset features, it would be nice if you had some thoughts on how to best organize presets so they are well suited for contribution, and at the same time, easy to maintain.

Like: 
- should we have one .ini for each vehicle type, or AP version ?
- a naming system for the .ini files depending on previous question.
- should the preset names start with something special, like type and version  "AP371:Baro vs GPSalt"  
- or maybe better "Baro vs GPSAlt , AP371"  (in case AP3.9.0 have some changes, those will be side by side ?)

Arne-W, 2017-05-04:
@AndKe thanks for the wishes - looks very good until now :smiley: 
But back to presets - I thought about one file for each vehicle type and for the presets I would prefer "Baro vs GPSAlt , AP371". The version is a addon to the name and should be only used if necessary. This is an easy and intuitive approach I think. If they really start growing we will have to think again - but then we will know the requirements to do it better.
But for now its quite good.

AndKe, 2017-05-07:
Could somebody make a build copy an default.ini somewhere to the release folder ?
-  then - on execution of apmplanner, if a new default.ini file (and old exists in ~/apmplanner2/), user is asked for approval to upgrade 
-if user denies, just copy the new default.ini as default.new to ~/apmplanner2/   - then user can, do a compare and merge any own settings.

Once we have that in place, I am ready to submit some presets.

Finally: there's 
For inspecting some settings VIBE.VibeX,Y,Z ,  and some others, at a glance, a specific scaling is needed.  for example; on   VIBE.Vibe*  must be within 60  so the graph scaled as 0..60 would always give a instant understanding of the vibrations.    - otherwise , with auto scale, even nice vibrations under 10 , may occupy the whole Y axis.   - and a catastrophally bad copter with vibrations at 40, will "look ok" if the graph is scaled to 150 due to one peak.

So: is to too much to ask to save/restore min/max limits too ?

Arne-W, 2017-05-07:
@AndKe At the moment it is not possible to have a group with min/max values. Groups always scale automatic. If you want a group with min/max you have to define it manually -> set all to manual scaling and give the same min/max values.

AndKe, 2017-05-07:
@Arne-W yes, and that is fine, (to edit every group member)  - but could min/max be saved & restored as a part of presets ?

Arne-W, 2017-05-07:
@AndKe They are saved and restored?! If a graph is not part of a group and is manual scaled the min/max are saved in the presets.

AndKe, 2017-05-07:
My bad, (for no reason), I did group the Vibe values, then discovered the
min Max was gone from the .ini file.