ArduPilot/apm_planner #1030: Graph: modes not correctly visualized.

AndKe, 2017-04-08:
Here's an ArduCopter log from an user I helped..
Auto, land  is not properly graphed, apparently AP2 only sees stabilize (mode0)

[20170328011956.log.zip](https://github.com/ArduPilot/apm_planner/files/907403/20170328011956.log.zip)
[image]

Arne-W, 2017-04-08:
This is an exported log - I don't know where it comes from but I have seen something like this before. The mode detection problem is caused by the log content as you can see in your own screen shot there is a string in the mode field, but the descriptor for mode messages describes an int8 followed by an int16.

AndKe, 2017-04-08:
by the few logged items, I would guess it is a .log from an APMx flight controller that saved .log , not a converted binary log, how can you tell for sure ? I asked the user, hoping he will tell which tool ,if any, were used.

AndKe, 2017-04-12:
I asked the person that provided the log aboute it's origin:
He replied:
"the log file was just directly downloaded from the APM 2.6"

It seems there must have been such isue around the time of AC3.2.1  (and most likely much older AC builds..)

Arne-W, 2017-04-14:
What software did he use for downloading? I am using ArduCopter since 3.1.5 and even my old logs have good mode messages. As this is no bin log and no tlog someone must have converted it.

AndKe, 2017-04-17:
I got this reply to the question:
"I just used mission planner on windows to download it via the log files tab. Not the telemetry one, the next one over."

Arne-W, 2017-04-17:
Hmm - OK - I have a suspicion that this is a bug in the Mission Planner when downloading the flashlogs. AFAIK the Mission Planner converts the flalshlog .bin files to this log format when downloading. Apparently it replaces the the mode ID with the interpreted text for the mode.
Nevertheless the format description for this message type does not fit the content.

AndKe, 2017-04-17:
it's an APM 2.6 , are you sure there even is a binary log format on it ?
I believe APM 2.6 stores text log files like earlier APM 2.x (with SD card) did , not a bin binary format. (I do not know if it's correct or not.)

AndKe, 2017-04-17:
I just "dug up" an old APM 2.5 I once used,  flashed with AC3.2.1  and armed for logging,(downloaded log with AP2 from master ) log is good.   
So you were right - it was caused by some bug in MP.

I just tried to help, and increase AP2 compatibility , personally I can barely understand why some would fly APM , not having EKF.  
Anyway - sorry for wasting your time.

closed 2017-04-17 by AndKe

Arne-W, 2017-04-17:
No problem you are always welcome with all sorts of questions. Especially if there could be a bug in the log analyzing code.