ArduPilot/apm_planner #1035: Map: panning with flightplan

AndKe, 2017-04-23:
@Arne-W  , please see https://youtu.be/qJW5WLXOkxg
Seems like pan/zoom is not trigging redraw

Arne-W, 2017-04-23:
I hate this waypoint stuff. I am not sure if we can get it work properly. Does this happen also when downloading waypoints from a FC?

AndKe, 2017-04-23:
tested:...Works just fine when connected to vehicle.  (both when loaded from plane, and from file..)
this reminds me of how the "mission elevation" widget (very useful) dont work unless connected to an flight controller. - same thing, regardless of mission being on the controller or not, it just don't work if not connected :(

Arne-W, 2017-04-25:
It was easier than I thought and until now it has no side effects....
If you want to test [https://github.com/Arne-W/apm_planner/tree/BugFixes](https://github.com/Arne-W/apm_planner/tree/BugFixes).
The PR will come...

AndKe, 2017-04-25:
works fine - thank you.

commit 97a6ee46 (2017-05-01): FlightPlan: Fix for waypoint lines not moving when panning map without having an UAV connected. Fixes #1035
closed 2017-04-25 by AndKe