ArduPilot/apm_planner #1036: KMZ export: plane icon orientation is incorrect

kd0aij, 2017-04-23:
the heading and RPY angles appear to update only on mode changes:
[image]

kd0aij, 2017-04-23:
It would also be nice to add Timestamp fields to allow animation and display of a subset of the data.

Arne-W, 2017-04-25:
As this feature was never implemented this is an enhancement. :smile: 
I made a small - slightly hacky - fix.
If you want to test use this branch: [https://github.com/Arne-W/apm_planner/tree/BugFixes](https://github.com/Arne-W/apm_planner/tree/BugFixes)
I will make a PR in the next days...

kd0aij, 2017-04-25:
Thanks for the fix; I wouldn't have known where to look...

? I didn't do anything to the plane model; is it a different size in your GE display?

Heading was correct, but roll and tilt needed a signflip: (no idea whether this patch uses QString properly, but it works)
```
diff --git a/src/output/kmlcreator.cc b/src/output/kmlcreator.cc
index 6baa473..c21bc65 100644
--- a/src/output/kmlcreator.cc
+++ b/src/output/kmlcreator.cc
@@ -524,8 +524,9 @@ void KMLCreator::writePlanePlacemarkElement(QXmlStreamWriter &writer, Placemark
 
                     writer.writeStartElement("Orientation");
                     writer.writeTextElement("heading", yaw);
-                        writer.writeTextElement("tilt", a.pitch());
-                        writer.writeTextElement("roll", a.roll());
+                    bool status;
+                        writer.writeTextElement("tilt", QString::number(-a.pitch().toFloat(&status)));
+                        writer.writeTextElement("roll", QString::number(-a.roll().toFloat(&status)));
                     writer.writeEndElement(); // Orientation
                 }
 
```
[image]

Arne-W, 2017-04-26:
Thanks for the info about roll and pitch - changed it :+1: 
The scaling of the plane is weird on my Ubuntu 16.04 with Google Earth 7.1.8.3036 (64-bit) it looks like this:
[image]

kd0aij, 2017-04-26:
Looks like the same version of OS and GE, but it will help to get timestamps working so you can reduce the clutter. I'm making progress on that; here's what mine looks like with timestamps added to the plane PlaceMarkers and only 1 second selected:
[image]

AndKe, 2017-04-26:
Google Earth Pro, 7.1.8.3036 (64-bit)
-please observe that the modes are for arducopter, not plane,  
- and yes, the plane symbol is huge, just like yours @Arne-W 

[image]

Arne-W, 2017-04-27:
Yes, at the moment only a message formatter for copters is used. Should not be a problem to fix that. The logananlysis class which triggers the export "knows" the MAV type so it could pass the type to the exporter. The KML exporter than just has to select the right message formatter.
@kd0aij perhaps you can have a look at this issue too when implementing the new time feature?

kd0aij, 2017-04-27:
I was wondering about that too; will take a look.

commit 3b299827 (2017-05-01): GraphView: Plane model in Gmaps now has correct yaw, pitch and roll values. Fixed corrupt CDATA structure. Fixes #1036
closed 2017-05-01 by billbonney