ArduPilot/apm_planner #1042: Cannot compile on Arch Linux

J-tt, 2017-05-15:
When compiling I get this error:

``` bash
src/ui/configuration/PX4FirmwareUploader.cc: In member function ‘bool PX4FirmwareUploader::checkCOA(const QByteArray&, const QByteArray&, const QString&)’:
src/ui/configuration/PX4FirmwareUploader.cc:644:136: error: invalid use of incomplete type ‘EVP_PKEY {aka struct evp_pkey_st}’
     int verify = RSA_verify(NID_sha1,(unsigned char*)serial.data(),serial.size(),(unsigned char*)signature.data(),signature.size(),pkey->pkey.rsa);
                                                                                                                                        ^~
In file included from /usr/include/openssl/crypto.h:31:0,
                 from /usr/include/openssl/comp.h:16,
                 from /usr/include/openssl/ssl.h:47,
                 from src/ui/configuration/PX4FirmwareUploader.cc:11:
/usr/include/openssl/ossl_typ.h:93:16: note: forward declaration of ‘EVP_PKEY {aka struct evp_pkey_st}’
 typedef struct evp_pkey_st EVP_PKEY;
                ^~~~~~~~~~~
make: *** [Makefile:69295: build-release/obj/PX4FirmwareUploader.o] Error 1
```

I'm building from the master branch, with everything up to date.

mayanez, 2017-05-16:
I just ran into the same issue and figured out a fix. Arch Linux seems to use both OpenSSL 1.0 and 1.1 and for some reason 1.1 doesn't have `evp_pkey_st` defined. If you add `INCLUDEPATH += /usr/include/openssl-1.0` and `LIBRARYPATH += /usr/lib/openssl-1.0` to `apm_planner.pro` it should compile without issue assuming all your other dependencies are correctly setup. I'll try and submit a PR once I've fully tested this out.

J-tt, 2017-05-19:
That worked for me, thank you! 👍

Arne-W, 2017-05-19:
Cool you found a solution :+1: 
It would be great if you could make a PR Fixing this issue!

J-tt, 2017-05-19:
@mayanez ^^

kikislater, 2017-05-19:
I link this discussion to aur package here : https://aur.archlinux.org/packages/apm_planner/

kikislater, 2017-05-25:
Hi,
Works well on Archlinux. Under another laptop, I have to adapt .pro file to get it working under Manjaro. 
```
    contains( DISTRO, "Manjaro" ) {
        message(Manjaro Build)
        DEFINES += Q_ARCHLINUX
        INCLUDEPATH += /usr/include/openssl-1.0
        LIBRARYPATH += /usr/lib/openssl-1.0
```

drobban, 2018-03-11:
Is this still an issue?

Arne-W, 2020-02-27:
Open SSL is not needed anymore. See c64ead17c7dc539d65d8bc9deb08284c315162e1

closed 2020-02-27 by Arne-W