ArduPilot/apm_planner #1044: Unable to change log directory

badaniels, 2017-05-22:
On APM Planner there is the option to set which directory the log files are saved to. However, you are unable to change these away from the default setting. Although the menu does allow to change the directory, it does not enable this change of directory and it does not save this setting for next time the app is launched. 

I am trying to change the tlogs directory to a dropbox so that I can automatically have my flight data saved to a shared folder.

pepak, 2017-06-24:
Same bug as [this](https://github.com/ArduPilot/apm_planner/issues/1002).

Arne-W, 2017-07-01:
@pepak thanks for testing.
@badaniels I am closing this issue cause its a duplicate and I am not able to reproduce it with the current master.

closed 2017-07-01 by Arne-W