ArduPilot/apm_planner #1045: linux build requires libqt5serialport5-dev and fails on debian anyway

marcmerlin, 2017-05-26:
I followed the build instructions:
saruman:~/RC/apm_planner$ qmake apm_planner.pro
Project MESSAGE: Qt version 5.7.1
Project MESSAGE: Linux build x64_86
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /home/merlin/RC/apm_planner DESTDIR /home/merlin/RC/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /home/merlin/RC/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View
Project ERROR: Unknown module(s) in QT: serialport

So I did
apt-get install libqt5serialport5 libqt5serialport5-dev
qmake worked, but the build failed:

ibs/alglib/src/linalg.cpp:26969:12: note: ‘estold’ was declared here
     double estold;
            ^
libs/alglib/src/linalg.cpp:27224:35: warning: ‘temp’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     rsave->ptr.p_double[3] = *temp;
                                   ^
libs/alglib/src/linalg.cpp:26971:12: note: ‘temp’ was declared here
     double temp;
            ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:48:0,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qmetaobject.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QMetaEnum:1,
                 from libs/opmapcontrol/src/core/maptype.h:30,
                 from libs/opmapcontrol/src/core/alllayersoftype.h:30,
                 from libs/opmapcontrol/src/core/alllayersoftype.cpp:27:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qhash.h: In member function ‘QPair<QHash<K, V>::iterator, QHash<K, V>::iterator> QHash<K, V>::equal_range(const Key&)’:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qhash.h:957:10: error: ‘pair’ does not name a type
     auto pair = qAsConst(*this).equal_range(akey);
          ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qhash.h:958:31: error: ‘pair’ was not declared in this scope
     return qMakePair(iterator(pair.first.i), iterator(pair.second.i));
                               ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qhash.h:958:31: note: suggested alternative:
In file included from /usr/include/c++/5/utility:70:0,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qcompilerdetection.h:1001,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:83,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qnamespace.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QMetaObject:1,
                 from libs/opmapcontrol/src/core/maptype.h:29,
                 from libs/opmapcontrol/src/core/alllayersoftype.h:30,
                 from libs/opmapcontrol/src/core/alllayersoftype.cpp:27:
/usr/include/c++/5/bits/stl_pair.h:96:12: note:   ‘std::pair’
     struct pair
            ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:56:0,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:51,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qmetaobject.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QMetaEnum:1,
                 from libs/opmapcontrol/src/core/maptype.h:30,
                 from libs/opmapcontrol/src/core/alllayersoftype.h:30,
                 from libs/opmapcontrol/src/core/alllayersoftype.cpp:27:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h: At global scope:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h:66:54: error: ‘>>’ should be ‘> >’ within a nested template argument list
     template <> struct TypesAreDeclaredMetaType<List<>> { enum { Value = true }; };
                                                      ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h:68:89: error: ‘>>’ should be ‘> >’ within a nested template argument list
     { enum { Value = QMetaTypeId2<Arg>::Defined && TypesAreDeclaredMetaType<List<Tail...>>::Value }; };
                                                                                         ^
Makefile:33374: recipe for target 'build-release/obj/alllayersoftype.o' failed
make: *** [build-release/obj/alllayersoftype.o] Error 1
make: *** Waiting for unfinished jobs....

marcmerlin, 2017-05-26:
Ok, finally 2h later, finally got a build :)

Arne-W, 2017-07-01:
@marcmerlin which Linux version are you using?? The openssl issue should be fixed through this #1043. It would be cool if you could verify it. I will add libqt5serialport5-dev and libqt5opengl5-dev to the Readme.

Arne-W, 2017-07-01:
@marcmerlin if #1043 does not fix the openssl issue could you please test if #1052 does?

marcmerlin, 2017-07-01:
@Arne-W  this was on debian/testing.
However I see your change was merged on May 24th while I tried to (and failed) to build on May 25th.
But looking at your fix, I'm not sure it would work for me. All you do is ensure that you're linking against an old openssl when you have both on your system.
On debian, you can't have 2 versions of a library's -dev package, so since apmplanner is incompatible with openssl 1.1, my problem is still there after your patch.
For reference on debian the includes are in /usr/include/openssl/ so you can't have 2 versions or point to an older version.

Arne-W, 2020-02-27:
Open SSL is not needed anymore. See c64ead17c7dc539d65d8bc9deb08284c315162e1

closed 2020-02-27 by Arne-W