ArduPilot/apm_planner #1047: FAILURE: Unsupported CMD: 400

KimEJ, 2017-05-31:
I followed the document to tried the motor arming.
[Http://ardupilot.org/copter/docs/arming_the_motors.html](url)
However, it did not work and I tried it by pressing the 'ARM' button on the 'ACTOINS' tab in the APM Planner.
If press the button, the following error occurs:

> [MAV 001:1] Calibrating barometer 
> [MAV 001:1] barometer calibration complete 
> [MAV 001:1] FAILURE: Unsupported CMD: 400 

Why do I get this error?
How can I solve this?

utkrishtsharma, 2019-05-14:
same error

Arne-W, 2019-05-18:
As far as I know this is no problem. The command informs the GCS about the UAV being disarmed / or armed. [See here](https://mavlink.io/en/messages/common.html#MAV_CMD_COMPONENT_ARM_DISARM). So nothing to worry about.