ArduPilot/apm_planner #1050: Graph: please check autorange

AndKe, 2017-06-15:
RCOU is ranged as -2.5 to max  - mostly due to "NONE" level in the beginning.
This reduces the resolution of the graph.  Can we just omit setting scale to "NONE" ?

[image]

Arne-W, 2017-06-15:
The "NONE" value is always shown if there is no measurement for the current cursor/mouse position, as soon as there is a value it will be shown.
There must be a measurement in one of the RCOU channels which has -2.5 as a value. The "NONE" value only means "there is no measurement" and will not be considered when calculating the min/max values.

AndKe, 2017-06-15:
please see log https://drive.google.com/file/d/0B9Rr1JTp37U9MV8zNjJJWTVQMnM/view?usp=sharing
there is no "-2.5" 
verified using filter + grid view, and mavlogdump + grep

Arne-W, 2017-06-16:
@AndKe Could not reproduce this issue. On my system the grouping dialog looks like this:
[image]
The min/max values which are used to determine the group min/max are the smallest and the biggest Axis values of all graphs is a group at the **moment** the "Advanced Scaling" button is pressed. So the min/max values can be changed by recaling some of the axis manually.
You could try to press the "reset scaling" button and after that open the scaling dialog.

AndKe, 2017-06-17:
sorry, I just realized it's caused by presets.
I've made some analyzing presets , to keep it simple (one preset for all kinds of multirotors), the RCOU1-8 are included.
it's caused by loading the preset for a hex, removing motor 7,8 does not help. 

it seems that a proper solution would be to make different presets for quad,hex,octo ?

Arne-W, 2017-06-17:
No problem :smile: 
As there is no way to detect whether the data is valid or not, different presets are the way to go I think.

closed 2017-06-17 by AndKe