ArduPilot/apm_planner #1058: Apmplanner creates the "mapscache" directory in home

pepak, 2017-07-02:
On Linux Apmplanner 2.0.25 creates the "mapscache" directory with "Data.qmdb" file in the home directory. Is this used? It creates a mess in the home.
Another "mapscache" directory with "Data.qmdb" file is in AppData location specified in the apmplanner configuration. Only this location should be used. Or better ".cache" directory in the home.

billbonney, 2017-07-02:
The offending line of code is here https://github.com/ArduPilot/apm_planner/blob/7f6ed47238a6e2994709177fc3af6fd5238fdc1a/libs/opmapcontrol/src/core/cache.cpp#L57

It initializes with home, then is moved here

https://github.com/ArduPilot/apm_planner/blob/38b21e1f0c4e9f4203f5a273aab4a57d1d3294ba/src/ui/map/QGCMapWidget.cc#L26

closed 2017-09-07 by billbonney

chocol4te, 2017-11-08:
Is there a reason this never got merged into master? I am still experiencing this issue.

Arne-W, 2017-11-08:
No - it is merged into master since Sep 7.
But there was no release since then - you have to build yourself from master

chocol4te, 2017-11-09:
Ah, thanks!

commit 95c95107 (2017-09-07): FlightPlan: Fixed creation of mapscache directory in home. Fixes #1058

The mapscache directory was initially created directly in home and then moved to apmplanner directory.
The directory directly in home is not created anymore.